SICKAG/sick_scan_xd #4: How does one use this without ROS

themightyoarfish, 2021-11-09:
The code compiles without ROS, but I see no way to use it in library mode – the wrappers define publishers, but don't expose functionality for "subscribing". How can I retrieve the data from a scanner when I use this project as a library?

michael1309, 2021-11-11:
Since there is no well-defined interface for a subscription in the IT world, we decided at this point to show as an example how the data can be prepared without ROS by exporting the data as a Birds Eye view as a jpg at intervals of a few seconds. You can find an example here in the code section https://github.com/SICKAG/sick_scan_xd/blob/0c35a36cc4cf40d2a04325e2145112c27623acd1/driver/src/sick_scan_common.cpp#L4215 In the current implementation, this data is updated in the directory ``` sick_scan_xd\demo``` directory. You can open the HTML file there in your default browser and inspect the lidar data and adjust the code section that exports the image data and CSV data to your needs. In principle it would also be possible to create a library with callback. But up to now, according to ours, it was easy for the users to branch the code and directly use the evaluation for performance reasons instead of the CSV export and the JPEG export.

regards michael

michael1309, 2022-01-19:
Closed due to inactivity.

closed 2022-01-19 by michael1309