SICKAG/sick_scan_xd #11: Image Viewer not showing simulation scan

Jiaxin-lim, 2021-12-09:
Hi, Would like to ask why is the image viewer not showing any scan results? i can't seems to fixed this even after successfully installing based on the steps

[image]

Jiaxin-lim, 2021-12-17:
Hi, any help in solving this issue is much appreciated. Have been trying over more than a week (changing IP address, ping IP address) and it is still unable to work

rostest, 2022-01-18:
Thanks for reporting and sorry for the late reply. It looks like there is a binary/ascii comunication problem: The driver expects binary Cola-B communication, while the lidar is using ascii Cola-A communication. It is recommended to use  binary Cola-B communication due to better compatibility and lower network traffic:
1. Set parameter "use_binary_protocol" to "true" in the launch file (which is the default), and
2. Set the lidar communication mode with the SOPAS ET software to "binary" and save this setting in the scanner's EEPROM.
Please update to the latest sick_scan_xd release (V2.4.3), too. This version has improved checking and switching to support binary communication.

closed 2022-01-19 by michael1309