SICKAG/sick_scan_xd #13: Lidar LMS511 connection problem

HenryOsnabrueck, 2022-01-18:
System: 
ros2 foxy
ubuntu 20.04 (kernal 5.11.0-46-generic)
sick_scan_xd 
LMS511-20100

Problem: When executing "ros2 run sick_scan sick_generic_caller ./src/sick_scan_xd/launch/sick_lms_5xx.launch" i get the following error: 
## ERROR SickScanMessages::parseLIDoutputstateMsg(): received 95 byte with invalid format (/home/henry/testbench_ws/src/sick_scan_xd/driver/src/sick_scan_messages.cpp:142): \x02\x02\x02\x02\x00\x00\x00VsSN LIDoutputstate 0 0 0 16 0 16 0 16 0 16 0 16 0 16 2 0 2 0 2 0 2 0 2 0 2 0 2 0 2 0 0w

Note: I changed the IP-address in the launch file according to the one shown in SOPAS. Also i set sw_pll_only_publish to false according to an error i got before.

Thanks in advance :) 

Full terminal code: 

[error.txt](https://github.com/SICKAG/sick_scan_xd/files/7888517/error.txt)

rostest, 2022-01-18:
Thanks for reporting. Unfortunately, we can't reproduce the error using a LMS511 on ROS-2 Foxy. The datagram "\x02\x02\x02\x02\x00\x00\x00VsSN LIDoutputstate 0 0 0 16 0 16 0 16 0 16 0 16 0 16 2 0 2 0 2 0 2 0 2 0 2 0 2 0 2 0 0" looks like an incorrect mix of binary and ascii telegram.
Please update to the latest sick_scan_xd release (V2.4.3) with improved switching to binary communication and check that both the driver and the lidar is running in binary communication mode:
1. Set parameter "use_binary_protocol" to "true" in the launch file (which is the default), and
2. Set the lidar communication mode with the SOPAS ET software to "binary" and save this setting in the scanner's EEPROM.

The following screenshot shows an example (LMS511 with latest sick_scan_xd on ROS-2 Foxy):
[image]

HenryOsnabrueck, 2022-01-19:
Thank you for the detailed and quick answer you gave me. 
After downloading the newest Version (V2.4.3) everything worked again, no more errors!
Also i did not have to change any more settings or parameters. Just straight up works again.

closed 2022-01-19 by HenryOsnabrueck