SICKAG/sick_scan_xd #14: LD-LRS36x1 min/max angle support

nigeno, 2022-01-18:
It appears like the driver doesn't support setting min/max angle scanning range for LD_LRS3601 scanner. sick_lrs_36x1.launch doesn't have this option yet as opposed to other older drivers. Are you planning on adding that in the nearest future?

nigeno, 2022-01-18:
Thank you for your prompt reply!
That is correct. Specifically, start/stop angle part of telegram.
[image]

rostest, 2022-01-25:
The latest release (V2.4.4) supports configuration of min and max angle for LRS36x1. They can now be configured in sick_lrs_36x1.launch:
```
        <param name="min_ang" type="double" value="-1.570796"/> <!-- default start angle for LRS 36x1: -90 deg -->
        <param name="max_ang" type="double" value="4.712389"/>  <!-- default stop angle for LRS 36x1: +270 deg -->
```

closed 2022-01-25 by rostest