SICKAG/sick_scan_xd #17: Unable to Connect to TIM781

JWhitleyWork, 2022-01-26:
I am attempting to connect to two different TIM781 lidars. If the lidars are configured for ASCII, I get errors about unexpected ASCII datagrams. However, if I configure the lidars to use binary and change the launch file config to binary, I get the following timeout on both sensors (excerpt from the end of the log because it produces a bunch of other data as well):

```
[INFO] [1643225657.236905141] [sick_scan]: Safety field 41, type 2 : (0,0), (-7.85398e-05,-1), (1.2491,-0.999872), (1.24957,0.999283), (0.000392588,1)
[INFO] [1643225657.236938302] [sick_scan]: Safety field 42, type 2 : (0,0), (-0.00011781,-1.5), (2.25021,-1.49939), (2.25092,1.49833), (0.000588882,1.5)
[INFO] [1643225657.236970942] [sick_scan]: Safety field 43, type 2 : (0,0), (-9.81748e-05,-1.25), (1.8749,-1.24931), (1.87549,1.24842), (0.000490735,1.25)
[INFO] [1643225657.237004782] [sick_scan]: Safety field 44, type 2 : (0,0), (-7.85398e-05,-1), (1.49959,-0.999223), (1.50006,0.998516), (0.000392588,1)
[INFO] [1643225657.237038062] [sick_scan]: Safety field 45, type 2 : (0,0), (-0.00011781,-1.5), (2.62647,-1.49673), (2.62717,1.49549), (0.000588882,1.5)
[INFO] [1643225657.237071502] [sick_scan]: Safety field 46, type 2 : (0,0), (-9.81748e-05,-1.25), (2.18858,-1.24719), (2.18917,1.24616), (0.000490735,1.25)
[INFO] [1643225657.237104542] [sick_scan]: Safety field 47, type 2 : (0,0), (-7.85398e-05,-1), (1.75069,-0.997653), (1.75116,0.996828), (0.000392588,1)
[INFO] [1643225657.237161662] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x01 0x00 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 CRC:<0x43>
[INFO] [1643225657.238867029] [sick_scan]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[INFO] [1643225657.238920949] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[INFO] [1643225657.240016553] [sick_scan]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x01\x01\x00\x00\x00\x00\x00\x00...
[INFO] [1643225657.240238474] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0018>sWN FREchoFilter 0x00 CRC:<0x7f>
[WARN] [1643225777.242375258] [sick_scan]: Timeout during waiting for new datagram
[WARN] [1643225777.242564218] [sick_scan]: sendSOPASCommand: no full reply available for read after 120000 ms
[INFO] [1643225777.242652619] [sick_scan]: Receiving: <STX><ETX>
[INFO] [1643225777.242696339] [sick_scan]: SOPAS Communication -Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0018>sWN FREchoFilter 0x00 CRC:<0x7f>

[INFO] [1643225777.242757379] [sick_scan]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[ERROR] [1643225777.242853259] [sick_scan]: ## ERROR in mainGenericLaser: init failed, retrying...
[INFO] [1643225777.242890539] [sick_scan]: Start initialising scanner [Ip: 192.168.0.109] [Port:2112]
[WARN] [1643225777.242924820] [sick_scan]: Disconnecting TCP-Connection.
[ERROR] [1643225777.243033860] [sick_scan]: Tcp::readInputData: Read 0 bytes, connection is lost!
[INFO] [1643225777.243194301] [sick_scan]: SickThread TcpRecvThread finished (flags: threadShouldRun=0, endThread=0).
sick_scan driver closed.
[ERROR] [1643225777.253615459] [sick_scan]: ## ERROR in mainGenericLaser: exception parameter 'diagnostic_updater.period' has already been declared
```

rostest, 2022-01-27:
Thanks for reporting! Did you use the latest sick_scan_xd release? This problem is solved with the most recent version of this driver (#15 also for TiM7xx). Please update sick_scan_xd to the latest version and retry. I'll close the ticket for now; but don't hesitate to reopen the issue in case the problem persists.

closed 2022-01-27 by rostest

JWhitleyWork, 2022-01-29:
Thank you! The latest version did indeed fix the problem.