SICKAG/sick_scan_xd #20: Unable to connect to NAV350

Nyamkani, 2022-02-14:
Hello!

I'm trying to connect to my NAV350 using ros2("sick_nav_3xx.launch"). but, I'm stuck in this error.

[image]

I tried this things :
Used latest "sick_scan_xd" driver and clean installed,
Changed Cola - ascii to binary (Using SOPAS ET.),
Changed the address ip (direct connection and router connection)

Any help would be appreciated!!

rostest, 2022-02-14:
Thanks for reporting. sick_scan_xd supports NAV310, but NAV350 is currently not officially supported. See https://github.com/SICKAG/sick_scan_xd#supported-hardware for a list of supported hardware devices.

Your configuration looks good, but the following error occures during the initial SOPAS startup sequence:
* Switch to userlevel "Authorized client" is successful (request "sMN SetAccessMode 0x03 ..." confirmed by "sAN SetAccessMode \x01")
* Command "sMN LMCstopmeas" fails due to wrong userlevel (NAV350 responds with errorcode 0x01, "sFA \x01" i.e. Sopas_Error_METHODIN_ACCESSDENIED, "Wrong userlevel, access to method not allowed"). The expected response to "sMN LMCstopmeas" is "sAN LMCstopmeas".

To support NAV350 devices, the sick_scan_xd driver needs to be adapted. We'll discuss possible time schedules internally and get back to this issue as soon as possible. In the meantime, the documents on https://www.sick.com/de/de/mess-und-detektionsloesungen/2d-lidar-sensoren/nav3xx/nav350-3232/p/p256041 and https://answers.ros.org/question/182215/ros-driver-for-sick-nav350/ might be useful.

Nyamkani, 2022-02-16:
I did factory default for my NAV350, But the result is same.
In fact. I thought that was just communication error. but I found this error is caused by not supporting specified NAV350 telegrams.

Thank you for kindly comment and explanation. I hope nav350 will be supported soon.

closed 2022-02-17 by Nyamkani

TillmanOttBA, 2022-03-07:
Hello,

we are planning to use the Sick Nav350 in Ros2 in the near future. When can we expect an integration?

Greetings
Tillman Ott

michael1309, 2022-03-07:
Hi Tillman,
we clarity this issue with the SICK support and management and give you a feedback until end of this week.