SICKAG/sick_scan_xd #21: How to use the python launch file format with a yaml config?

Kaju-Bubanja, 2022-02-23:
I am migrating from sick_scan2 to sick_scan_xd. Thanks for this repo and the collected and nicely presented information. I have following launch file which worked with sick_scan2's sick_generic_caller, but doesn't with the new one.

```
import sys
import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    base_config_path = os.path.join(get_package_share_directory('my_package'), 'config')
    sick_config = os.path.join(base_config_path, "sensor", "lms_xd.yaml")
    return LaunchDescription([
        Node(
            package='sick_scan',
            executable='sick_generic_caller',
            output="screen",
            parameters=[sick_config],
        )
    ])
```
and this yaml file:

```
sick_scan:
  ros__parameters:
    hostname : "192.168.254.26"

    # Default values from the sick_scan_xd/launch/sick_lms5xx.launch file
    frame_id: "cloud"
    scanner_name: "sick_lms_5xx"
    port: 2112
    min_ang: -1.658062789
    max_ang: 1.658062789
    scanner_type: "sick_lms_5xx"
    range_max: 100.0
    intensity: true
    intensity_resolution_16bit: false
    timelimit: 5
    use_binary_protocol: true
    sw_pll_only_publish: true
    min_intensity: 0.0
    start_services: true
    activate_lferec: true
    activate_lidoutputstate: true
    activate_lidinputstate: true
```

when launching my launch file I get following stacktrace:

```
[INFO] [sick_generic_caller-1]: process started with pid [959602]
[sick_generic_caller-1] [INFO] [1645611240.611344578] [sick_scan]: sick_generic_caller V. 2.4.2
[sick_generic_caller-1] 
[sick_generic_caller-1] [INFO] [1645611240.612554789] [sick_scan]: Program argument 1: /home/user/ros2_rolling/install/sick_scan/lib/sick_scan/sick_generic_caller
[sick_generic_caller-1] 
[sick_generic_caller-1] [INFO] [1645611240.613560024] [sick_scan]: Program argument 2: --ros-args
[sick_generic_caller-1] 
[sick_generic_caller-1] [INFO] [1645611240.614298742] [sick_scan]: Program argument 3: --params-file
[sick_generic_caller-1] 
[sick_generic_caller-1] [INFO] [1645611240.615034321] [sick_scan]: Program argument 4: /home/user/ros2_rolling/install/my_package/share/my_package/config/sensor/lms_xd.yaml
[sick_generic_caller-1] 
[sick_generic_caller-1] [ERROR] [1645611240.615913789] [sick_scan]: ## ERROR parseLaunchfileSetParameter(): Tag-Value setting not valid. Use pattern: <tag>:=<value>  (e.g. hostname:=192.168.0.4) (Check the entry: --ros-args)
[sick_generic_caller-1] 
[sick_generic_caller-1] [ERROR] [1645611240.615967060] [sick_scan]: ## ERROR sick_generic_laser: parseLaunchfileSetParameter() failed, aborting
[sick_generic_caller-1] 
[ERROR] [sick_generic_caller-1]: process has died [pid 959602, exit code 255, cmd '/home/user/ros2_rolling/install/sick_scan/lib/sick_scan/sick_generic_caller --ros-args --params-file /home/user/ros2_rolling/install/my_package/share/my_package/config/sensor/lms_xd.yaml'].
not found: "/home/user/ros2_rolling/install/rmw_cyclonedds_cpp/share/rmw_cyclonedds_cpp/local_setup.bash"
```

Now I looked at the source code and I'm not sure what the function: parseLaunchfileSetParameter under sick_generic_laser.cpp is trying to parse. Is it expecting an xml launch file? Or can it handle yaml, but somehow it trips and tries to parse `--ros-args` instead of the actual file?

I would like to keep my current setup since I do more things in the python launch file and it is more flexible and I couldn't do those things with an old style xml launch file. Is this possible with the sick_generic_caller in this repo?

rostest, 2022-02-24:
Thanks for your kind report. To be compatible with ROS-1, ROS-2 and non-ROS-environments, the driver parses the xml-launchfile to read its configuration. A yaml-configuration is currently not supported. We will discuss internally about a possible yaml-configuration. In the meantime, the comments in [#18](https://github.com/SICKAG/sick_scan_xd/issues/18) (Run sick_scan from launch-file) can be helpful.

rostest, 2022-02-28:
sick_scan_xd has now been updated. Release 2.4.5 includes python-launchfiles for ROS-2. Please update the sources and rebuild. You can call sick_generic_caller from python e.g. by

```
import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    ld = LaunchDescription()
    sick_scan_pkg_prefix = get_package_share_directory('sick_scan')
    launch_file_path = os.path.join(sick_scan_pkg_prefix, 'launch/launch/sick_lms_5xx.launch')
    node = Node(
            package='sick_scan',
            executable='sick_generic_caller',
            output='screen',
            arguments=[launch_file_path]
    )    
    ld.add_action(node)
    return ld
```

To overwrite optional parameter like hostname, just use `arguments=[launch_file_path, "hostname:=192.168.0.123"]`.

We decided to keep the launchfile configuration for max. compatibility with ROS-1, ROS-2 and non-ROS-systems.

@JWhitleyWork Thank you very much for help and support!

closed 2022-02-28 by rostest