SICKAG/sick_scan_xd #24: Stop the Scanner

issambm, 2022-03-05:
How to stop the scanner after executing for example:
`roslaunch sick_scan sick_nav_3xx.launch hostname:=<ip-address>`

michael1309, 2022-03-06:
There are at least two methods to stop the lidar device driver:
* Use <STRG>+c to stop it manually
* Use the following tool to stop the driver [kill_generic_caller](https://github.com/SICKAG/sick_scan_xd/tree/master/tools/kill_generic_caller.sh)

issambm, 2022-03-06:
Thank you for your reply.

I am actually talking about stopping the physical rotation of the scanner after your do `Ctrl+C`. Would you please let me know how to do that?

issambm, 2022-03-07:
I was wondering if there is even a way to stop the rotation of the scanner via command-line, other than doing this through SOPAS ET software every time which is inconvenient.

Thanks in advance for any help you could provide on this.

rostest, 2022-03-08:
Thanks for reporting this issue. A driver update is available on the branch https://github.com/SICKAG/sick_scan_xd/tree/feature/lidar_stop_exit. This driver update supports a new ros service to stop the scanner at exit. Please checkout the branch and rebuild. 

To stop the scanner rotation and to exit the driver, you can call the new ros service `SickScanExit` by running
```
rosservice call /sick_nav_3xx/SickScanExit "{}" # stop scanner and exit on ROS-1
ros2 service call /SickScanExit sick_scan/srv/SickScanExitSrv "{}" # stop scanner and exit on ROS-2
```
Alternatively you can use Ctrl-C, which now stops the scanner and exits the driver, too.

Please leave a short message, so that we can merge the branch to the main trunk.

issambm, 2022-03-08:
Great, thank you!