SICKAG/sick_scan_xd #28: Invalid angles for lrs_4xxx

robertwil, 2022-03-16:
Hi,
while trying to get scan data from the LRS4000  using the sick_lrs_4xxx.launch I got the following error

` Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., received answer: "sFA\x00\x04", expected patterns: "sWA LMPoutputRange","sAN LMPoutputRange`

an the node failed.
Digging a little bit in the telegram listing and the documentation I found that the angles are restricted to [-180°, 180°] which would also fit the error message (sFA\x00\x04 = Sopas_Error_LOCALCONDITIONFAILED ?) as the angles are set to [-270°,90°] in the launch:

```
       <param name="min_ang" type="double" value="-4.71238898"/> <!-- value from sick_scan2/config/sick_lrs_4xxx.yaml -->
        <param name="max_ang" type="double" value="1.57079632"/>  <!-- value from sick_scan2/config/sick_lrs_4xxx.yaml -->
```

Setting the limit to [-180°,180°] everything seems to work as expected.
I'm not sure if it's enough to just change the values in the launch file or if there should be additional checks/calculations in the (ROS) driver. 
Another artifact that might result from this issue is that the published cloud and the lasercan have a different "origin" and do therefore not match. I guess it is due to substracting 90° from the startAngle in sick_Scan_common.cpp line 4154
`                              msg.angle_min = startAngle / 180.0 * M_PI - M_PI / 2;`

Is this done to have the scan published from basically  0° (-360°) to 0° and "fails" here because we can't set [-270°,90°] ?

I'm using the driver with ROS1 noetic and compiled it as it is described under the respective chapter.

rostest, 2022-03-16:
Thanks for reporting! Min/max angles [-180°,180°] are correct for the LRS-4xxx, but were erroneously configured in the launchfile. The correct settings for LRS-4xxx are
```
        <param name="min_ang" type="double" value="-3.1415926"/>
        <param name="max_ang" type="double" value="3.1415926"/> 
```
This is enough to correct the failure (no further angle offset applies). Please use these settings or checkout the latest driver versions (released just now), which contains the correct LRS-4xxx settings.

robertwil, 2022-03-17:
Is there any convenient way (except from introducing another link) to have the scan published with the same orientation (so x-axis is the front of the scanner) as the cloud and like it is visible in the SICK tools (e.g. Sopas Air) ?

michael1309, 2022-03-17:
The current state of sick_scan_xd [V. 2.5.1] and newer supports your requested axis orientation for the LRS4xxx.
The X-axis points forward (0 degrees is the "red" X-axis in rviz). The connectors on the device are therefore in negative X-axis direction. The Y-axis points to the left in the ROS logic ("green" Y-axis in rviz). 
We consistently follow the ROS coordinate logic in the driver for all supported devices.
In the SICK lidars, the convention is often different (namely X-axis to the right and not in direction of the lidar face). Therefore tools from the company SICK may display a different axis orientation.

As you already wrote yourself, you can of course use a ROS static_transform_publisher to set the arbitrary axis orientations.

robertwil, 2022-03-17:
Hi, thanks for the quick response.
I hope I did not misunderstand you but looking at our setup

Screen in front of scanner:
[image]

The pointcloud from sick_scan_xd:
[image]

The scan from sick_scan_xd:
[image]

The scan in sopas air:
[image]

it does not seem that the conventions you mentioned apply here for the laserscan.
I would expect the screen to be visible in the laserscan directly in front of the X-axis (like in the cloud or sopas air).

I compiled and tested the most recent version.

michael1309, 2022-03-17:
Thanks for your explanatory pictures. Now your problem is clear. The Pointcloud message is correct. That is what our comments were referring to. 

We strongly recommend to use the Pointcloud messages. 
We support the Scan messages only for historical reasons.

michael1309, 2022-03-17:
Do you need the scan messages in your application?

rostest, 2022-03-17:
The error in the laserscan messages is fixed in the branch https://github.com/SICKAG/sick_scan_xd/tree/feature/lrs_4xxx_angles. 
You're right: As you already noticed, line 4154 in sick_scan_common.cpp causes an additional rotation of the laserscan messages. This is correct for most lidars, but not for LRS-4xxx. Replacing the fixed offset by the lidar-specific angle offset solves the issue:
`msg.angle_min = startAngle / 180.0 * M_PI + this->parser_->getCurrentParamPtr()->getScanAngleShift();`
Please checkout the branch https://github.com/SICKAG/sick_scan_xd/tree/feature/lrs_4xxx_angles and rebuild. Please leave a short note, so we can merge the branch into the master trunk.
Thanks again!

robertwil, 2022-03-18:
Hi, thanks for the quick fix.
I will be able to test it start of next week when I'm on site again.

Another thing I found is that the laserscan msg (not cloud) is published at the doubled frequency. As 50Hz (I've set 25Hz) is not even supported by the scanner I checked the messages and found that always two consecutive msgs have the exact same timestamp. In the code I did not find anything that could lead to that (except maybe this DiagnosticPublisher, but it seemed to be deactivated and I'm not sure if it would publish the laserscan msg like the normal publisher). Can you reproduce this behavior ? I could also create a additional issue for this if you prefer to have them split.

rostest, 2022-03-21:
Hi, thanks again for reporting this issue. The bug is fixed in the branch https://github.com/SICKAG/sick_scan_xd/tree/feature/lrs_4xxx_angles. Please update and rebuild.
As you've completely correctly assumed, the laserscan messages have been published twice, once by the default ros publisher, and once again by the diagnostic publisher. To fix the error, Llines 4520-4527 in sick_scan_common.cpp are replaced  by
```
                  // rosPublish(pub_, msg);
#if defined USE_DIAGNOSTIC_UPDATER // && __ROS_VERSION == 1
                  // if(diagnostics_)
                  //   diagnostics_->broadcast(diagnostic_msgs_DiagnosticStatus_OK, "SickScanCommon running, no error");
                  if(diagnosticPub_)
                    diagnosticPub_->publish(msg);
                  else
                    rosPublish(pub_, msg);
#else
                  rosPublish(pub_, msg);
#endif
```
Please update from the branch and retry. Thanks!

closed 2022-03-22 by rostest

rostest, 2022-03-22:
@robertwil Thanks for your support!

commit 7eb5a5e6 (2022-03-21): Fix #28 (duplicated laserscan messages)