SICKAG/sick_scan_xd #34: USB support

Noel215, 2022-04-06:
Hi!

I am working on the migration to ros2 of a robot which has the scanner connected by USB. 
For ros1 is used [sick_tim_common_usb.cpp](https://github.com/uos/sick_tim/blob/melodic/src/sick_tim_common_usb.cpp) but here and  in [sick_scan2](https://github.com/SICKAG/sick_scan2) package I haven't found any references to USB support.
Am I missing something?

rostest, 2022-04-06:
Hi,
USB connections are currently not supported. You can find further information in https://github.com/SICKAG/sick_scan2/issues/31 and https://github.com/SICKAG/sick_scan2/issues/22.

Noel215, 2022-12-29:
Hi again!

Since eventually I will need the USB support, I would like it to be implemented and supported. I could try it but I do not have so much idea of how it works and how to start so, if you could give me some hint, point of start or advice it would be great!

I am aware of the risk of loosing messages through the USB interface, as said in [SICKAG/sick_scan2#22](https://github.com/SICKAG/sick_scan2/issues/22)

Thanks!