SICKAG/sick_scan_xd #35: Change min_ang and max_ang failed (sick lms1xx)

NekoSaiKou, 2022-04-08:
Hi,
I have two LMS100 and I want to change the min_ang and max_ang using the provided [launch file](https://github.com/SICKAG/sick_scan_xd/blob/master/launch/sick_lms_1xx.launch)

However, I got this output

```
[ INFO] [1649388783.031327939]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[ INFO] [1649388783.032709519]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x13\x88\xff\xf9\x22\x30...
[ INFO] [1649388783.032934463]: Angle resolution of scanner is 0.5 [deg]  (in 1/10000th deg: 5000)
[ INFO] [1649388783.033030455]: [From:To] -45 [deg] to 225f [deg] (in 1/10000th deg: from -450000 to 2250000)
[ INFO] [1649388783.033108844]: MIN_ANG: -2.35619 [rad] -135 [deg]
[ INFO] [1649388783.033201720]: MAX_ANG: 2.35619 [rad] 135 [deg]
[ INFO] [1649388783.033326676]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x13 0x88 0xff 0xf9 0x22 0x33 0x00 0x22 0x55 0x0d CRC:<0x8c>
[ INFO] [1649388783.034770660]: Receiving: <STX>sFA\x00\x0a<ETX>
[ WARN] [1649388783.034928104]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., received answer: "sFA\x00\x0a", expected patterns: "sWA LMPoutputRange","sAN LMPoutputRange"
[ WARN] [1649388788.035193019]: Timeout during waiting for new datagram
[ WARN] [1649388788.035376670]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., received answer: "", expected patterns: "sWA LMPoutputRange","sAN LMPoutputRange"
[ERROR] [1649388788.035469570]: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., giving up after 1 unexpected answers.
[ INFO] [1649388788.035572149]: Failed to init scanner Error Code: 1
Waiting for timeout...
```

After that, it times out.
AFAIK, sFA\x00\x0a means Sopas_Error_VARIABLE_WRITE_ACCESSDENIED, and the document says that ```It is not allowed to write values to this variable. Probably the variable is defined as read-only.``` That is pretty weird.

I have read similar issue [here](https://github.com/SICKAG/sick_scan/issues/4), but it seems that the password is correct. Here is the full log

```
SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /sick_lms_1xx/activate_lferec: True
 * /sick_lms_1xx/activate_lidinputstate: True
 * /sick_lms_1xx/activate_lidoutputstate: True
 * /sick_lms_1xx/ang_res: 0.5
 * /sick_lms_1xx/client_authorization_pw: F4724744
 * /sick_lms_1xx/frame_id: cloud
 * /sick_lms_1xx/hostname: 192.168.51.3
 * /sick_lms_1xx/intensity: False
 * /sick_lms_1xx/intensity_resolution_16bit: False
 * /sick_lms_1xx/max_ang: 2.35619
 * /sick_lms_1xx/message_monitoring_enabled: True
 * /sick_lms_1xx/min_ang: -2.35619
 * /sick_lms_1xx/min_intensity: 0.0
 * /sick_lms_1xx/port: 2112
 * /sick_lms_1xx/range_max: 25.0
 * /sick_lms_1xx/read_timeout_millisec_default: 5000
 * /sick_lms_1xx/read_timeout_millisec_kill_node: 150000
 * /sick_lms_1xx/read_timeout_millisec_startup: 120000
 * /sick_lms_1xx/scan_freq: 50.0
 * /sick_lms_1xx/scanner_type: sick_lms_1xx
 * /sick_lms_1xx/start_services: True
 * /sick_lms_1xx/sw_pll_only_publish: True
 * /sick_lms_1xx/timelimit: 5
 * /sick_lms_1xx/use_binary_protocol: True

NODES
  /
    sick_lms_1xx (sick_scan/sick_generic_caller)

ROS_MASTER_URI=http://localhost:11311

process[sick_lms_1xx-1]: started with pid [571531]
[ INFO] [1649388781.560193350]: sick_generic_caller V. 2.6.1

[ INFO] [1649388781.561245937]: Program argument 1: /home/aiRobots/aiRobots/aiRobots_AMMR/Software/devel/lib/sick_scan/sick_generic_caller

[... 101 lines cut ...]
[ INFO] [1649388789.538667472]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x13\x88\xff\xf9\x22\x30...
[ INFO] [1649388789.538736747]: Angle resolution of scanner is 0.5 [deg]  (in 1/10000th deg: 5000)
[ INFO] [1649388789.538759644]: [From:To] -45 [deg] to 225f [deg] (in 1/10000th deg: from -450000 to 2250000)
[ INFO] [1649388789.538772916]: MIN_ANG: -2.35619 [rad] -135 [deg]
[ INFO] [1649388789.538784222]: MAX_ANG: 2.35619 [rad] 135 [deg]
[ INFO] [1649388789.538805645]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x13 0x88 0xff 0xf9 0x22 0x33 0x00 0x22 0x55 0x0d CRC:<0x8c>
[ INFO] [1649388789.540109536]: Receiving: <STX>sFA\x00\x0a<ETX>
[ WARN] [1649388789.540168796]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., received answer: "sFA\x00\x0a", expected patterns: "sWA LMPoutputRange","sAN LMPoutputRange"
[ WARN] [1649388794.540475245]: Timeout during waiting for new datagram
[ WARN] [1649388794.540610473]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., received answer: "", expected patterns: "sWA LMPoutputRange","sAN LMPoutputRange"
[ERROR] [1649388794.540690340]: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., giving up after 1 unexpected answers.
[ INFO] [1649388794.540794032]: Failed to init scanner Error Code: 1
Waiting for timeout...
```

I finally kill the process because the software tries to initialize the scanner again.

BTW, I have tried your old repository [sick_scan](https://github.com/SICKAG/sick_scan). I still cannot change the min_ang and max_ang. 
Parameter I used:
```
<param name="min_ang" type="double" value="0.0"/> 
<param name="max_ang" type="double" value="3.141"/> 
```

I got same ```SFA 00 0A``` error. The scanner is initialized successfully, but the output angle range **unchanged**.

NekoSaiKou, 2022-04-08:
Hi,

Thanks for your reply.
After factory reset, the LED indicates that everything is OK. But both of the scanner still cannot be set with SOPAS ET.  

Since these scanners had not been used for years, I wonder if the device version is too old
The device version:
* V1.20-14.11.2008
* V1.23-09.06.2009

Should I update the firmware?

[image]

rostest, 2022-04-08:
Thanks for your reply. Please use the most recent firmware and sick_scan_xd release and retry.

NekoSaiKou, 2022-04-10:
Sorry for my late reply.

I have trouble downloading firmware. "Download Firmware" button is disabled in the SOPAS ET window.
I'll contact SICK company in my country for help. Also, I would greatly appreciate if if anyone can tell me whether there is any official firmware update instruction or not.
If I have any progress on this issue, I will leave a comment ASAP.

michael1309, 2022-04-11:
We recommend, as you suggest yourself, to ask the official SICK support. You will get professional help updating your firmware there.

NekoSaiKou, 2022-04-13:
* Good news 
 This driver is OK. The driver works good on my LMS100 v1.60.
* Bad news
 SICK technical support told me that device version 1.20 (2008) and 1.23 (2009) are too old. The firmware cannot be modified. Therefore, I can only config the output angle range by post-processing.

closed 2022-04-21 by rostest