SICKAG/sick_scan_xd #47: Timestamps laserscan messages

rostest, 2022-05-04:
The timestamps in the header of pointcloud messages (i.e. msg.header.stamp of ros::sensor_msgs::PointCloud2 messages) are lidar ticks converted to ros-time using a software pll. The timestamp in the header of laserscan messages (msg.header.stamp of ros::sensor_msgs::LaserScan messages) ist the ros-time when receive the telegram. The timestamp of the laserscan messages should be identical to the timestamp of pointcloud messages.

rostest, 2022-05-04:
Release 2.6.3: Timestamp of pointcloud and laserscan messages identical and computed from lidar ticks by software-pll

closed 2022-05-04 by rostest

michael1309, 2023-07-31:
The so-called software PLL (see https://github.com/SICKAG/sick_scan_xd/blob/master/doc/software_pll.md ) is used to compare the so-called ticks of the lidar for the scan start against the PC system time. The lidars run asynchronously against each other, so that the timestamp at which a certain angle is reached differs between the lidars. For very fast movements, it is therefore advisable to determine a separate time stamp for each shot (angular direction) by using the start timestamp of the scan and then taking the rotation speed of the lidar into account.

Roughly:
```
Time stamp for a specific angle = timestamp at scan start + (angle difference to start angle)/360 [deg] * 1.0/scanning frequency
```