SICKAG/sick_scan_xd #49: free(): invalid pointer

Barkodcz, 2022-05-04:
Hi,

I'm trying to get some data from LMS 511-12100S06 sensor.
But unfortunately I have a crash :(
I tried on windows (10) and also on ubuntu (21.10-64 bit), it has the same result.

I'm looking at the backtrace and I think the problem is it has the same invalid free up.

My backtrace (for all threads):
```
(gdb) thread apply all bt                                                                                
                                                    
Thread 3 (Thread 0x7ffff7114640 (LWP 786276) "sick_generic_ca"):        
#0  0x00007ffff7b72d3f in __GI___poll (fds=0x7ffff7111b38, nfds=1, timeout=1000) at ../sysdeps/unix/sysv/linux/poll.c:29                                                                                           
#1  0x0000555555635d1c in poll (__timeout=<optimized out>, __nfds=<optimized out>, __fds=<optimized out>) at /usr/include/x86_64-linux-gnu/bits/poll2.h:48                                                         
#2  Tcp::readInputData (this=0x7ffff711b5f8) at /home/bbalazs/dev/projects/kozut/sick_scan_xd/driver/src/tcp/tcp.cpp:316                                                                                           
#3  0x0000555555638bd6 in Tcp::readThreadFunction (waitTimeMs=<optimized out>, endThread=<optimized out>, this=<optimized out>) at /home/bbalazs/dev/projects/kozut/sick_scan_xd/driver/src/tcp/tcp.cpp:262        
#4  SickThread<Tcp, &Tcp::readThreadFunction>::thread_entry (this=0x7ffff0004770) at /home/bbalazs/dev/projects/kozut/sick_scan_xd/include/sick_scan/tcp/SickThread.hpp:103                                        
#5  0x000055555564883d in wrapper_prerun (state=<optimized out>) at /home/bbalazs/dev/projects/kozut/sick_scan_xd/driver/src/tcp/SickThread.cpp:9                                                                  
#6  0x00007ffff7e5b694 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6                                
#7  0x00007ffff7aef947 in start_thread (arg=<optimized out>) at pthread_create.c:435
#8  0x00007ffff7b7fa44 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:100                      
                                                                                                         
Thread 2 (Thread 0x7ffff7a07640 (LWP 786274) "sick_generic_ca"):                                         
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737347876416) at pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737347876416) at pthread_kill.c:80                     
#2  __GI___pthread_kill (threadid=140737347876416, signo=signo@entry=6) at pthread_kill.c:91
#3  0x00007ffff7a9d476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26                    
#4  0x00007ffff7a837b7 in __GI_abort () at abort.c:79                        
#5  0x00007ffff7ae4606 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7ffff7c3613d "%s\n") at ../sysdeps/posix/libc_fatal.c:155                                                                  
#6  0x00007ffff7afbafc in malloc_printerr (str=str@entry=0x7ffff7c33d86 "free(): invalid pointer") at malloc.c:5543                                                                                                
#7  0x00007ffff7afd434 in _int_free (av=<optimized out>, p=<optimized out>, have_lock=0) at malloc.c:4326
#8  0x00007ffff7affd05 in __GI___libc_free (mem=<optimized out>) at malloc.c:3278                                                                                                                                  
#9  0x00005555555da2de in __gnu_cxx::new_allocator<float>::deallocate (__t=<optimized out>, __p=<optimized out>, this=<optimized out>) at /usr/include/c++/11/ext/new_allocator.h:145
#10 std::allocator_traits<std::allocator<float> >::deallocate (__n=<optimized out>, __p=<optimized out>, __a=...) at /usr/include/c++/11/bits/alloc_traits.h:492                                                   
#11 std::_Vector_base<float, std::allocator<float> >::_M_deallocate (__n=<optimized out>, __p=<optimized out>, this=<optimized out>) at /usr/include/c++/11/bits/stl_vector.h:354                                  
#12 std::_Vector_base<float, std::allocator<float> >::~_Vector_base (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/stl_vector.h:335                                                 
#13 std::vector<float, std::allocator<float> >::~vector (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/stl_vector.h:683                                                             
#14 sick_scan::SickScanCommon::loopOnce (this=0x7ffff7115010, nh=<optimized out>) at /home/bbalazs/dev/projects/kozut/sick_scan_xd/driver/src/sick_scan_common.cpp:4901                                            
#15 0x000055555559e9b2 in mainGenericLaserInternal (argc=<optimized out>, argv=<optimized out>, nodeName=..., nhPriv=<optimized out>, do_ros_spin=<optimized out>, exit_code=<optimized out>) at /home/bbalazs/dev/projects/kozut/sick_scan_xd/driver/src/sick_generic_laser.cpp:594          
#16 0x00005555555a0893 in GenericLaserCallable::mainGenericLaserCb (this=0x5555556d3bd0) at /home/bbalazs/dev/projects/kozut/sick_scan_xd/driver/src/sick_generic_laser.cpp:120                                    
#17 0x00007ffff7e5b694 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6          
#18 0x00007ffff7aef947 in start_thread (arg=<optimized out>) at pthread_create.c:435
#19 0x00007ffff7b7fa44 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:100
                                                    
Thread 1 (Thread 0x7ffff7a5a340 (LWP 786251) "sick_generic_ca"):            
#0  0x00007ffff7b3fa68 in __GI___clock_nanosleep (clock_id=clock_id@entry=0, flags=flags@entry=0, req=req@entry=0x7fffffffc620, rem=rem@entry=0x0) at ../sysdeps/unix/sysv/linux/clock_nanosleep.c:78              
#1  0x00007ffff7b44977 in __GI___nanosleep (req=req@entry=0x7fffffffc620, rem=rem@entry=0x0) at ../sysdeps/unix/sysv/linux/nanosleep.c:25                                                                          
#2  0x00007ffff7b75fe9 in usleep (useconds=<optimized out>) at ../sysdeps/posix/usleep.c:32              
#3  0x00005555555b579a in ros::spinOnce () at /home/bbalazs/dev/projects/kozut/sick_scan_xd/roswrap/src/rossimu/kinetic/src/rossimu.cpp:257                                                                        
#4  ros::spin () at /home/bbalazs/dev/projects/kozut/sick_scan_xd/roswrap/src/rossimu/kinetic/src/rossimu.cpp:264                                                                                                  
#5  0x000055555559736f in rosSpin (nh=<optimized out>) at /home/bbalazs/dev/projects/kozut/sick_scan_xd/include/sick_scan/sick_ros_wrapper.h:180                                                                   
#6  main (argc=0, argv=<optimized out>) at /home/bbalazs/dev/projects/kozut/sick_scan_xd/driver/src/sick_generic_caller.cpp:185  
```

and that's my output:
```
(gdb) r                                                                                                                                                                                                            
Starting program: /home/bbalazs/dev/projects/kozut/sick_scan_xd/build_linux/sick_generic_caller ../launch/sick_lms_5xx.launch                                                                                      
[Thread debugging using libthread_db enabled]                                                                                                                                                                      
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".                                                                                                                                         
[Info]: sick_generic_caller V. 2.6.3                                                                                                                                                                               
                                                                                                                                                                                                                   
[Info]: Program argument 1: /home/bbalazs/dev/projects/kozut/sick_scan_xd/build_linux/sick_generic_caller                                                                                                                             [... cut ...]
        <param name="message_monitoring_enabled" type="bool" value="True" />      
        <param name="read_timeout_millisec_default" type="int" value="5000"/>     
        <param name="read_timeout_millisec_startup" type="int" value="120000"/>   
        <param name="read_timeout_millisec_kill_node" type="int" value="150000"/> 
        <param name="client_authorization_pw" type="string" value="F4724744"/>    

    </node>
</launch>
```

Do you have any idea what the problem might be?

thank you in advance,
Best regards
Balázs

rostest, 2022-05-05:
Thanks for reporting! We will investigate this issue and get back to you soon.

rostest, 2022-05-05:
Hi,

unfortunately I can not reproduce the error, neither on Linux or Windows. Can you record the network-traffic with Wireshark until the error occures? With a pcapng-file we can replay the tcp-packages from the lidar and try to recover the error.
If possible:
* start Wireshark, 
* select the interface with the tcp-session to the lidar, 
* record the tcp data and
*  after the driver crashed, save the recorded to a pcapng-file. 

Please provide us with the pcapng-file for further diagnosis.
Many thanks in advance!

Barkodcz, 2022-05-06:
Hi,

uploaded the pcapng-file to this site https://files.fm/u/cpnydb88h

The LiDaR IP is 192.168.20.1 and the PC IP is 192.168.20.10

thanks in advance!

closed 2022-05-09 by rostest

rostest, 2022-05-09:
@Barkodcz Thanks to your feedback and pcapng-file we could analyze and fix the error. Due to erroneous echo filter settings, the number of received echos exceeded the number of expected echos, which caused a memory overwrite.

Please update to the latest release 2.6.4 and rebuild. Note that the number of echos can now be configured in the launchfile sick_lms_5xx.launch by
```
<param name="filter_echos" type="int" value="1"/> <!-- FREchoFilter settings: 0: first echo, 1: all echos, 2: last echo -->
```

Many thanks again for feedback and support!

Barkodcz, 2022-05-09:
Thank you for the support, now it is working :D