SICKAG/sick_scan_xd #62: LMS_5xx  -- Unable to set ang_res and scan_freq parameters

visiont3lab, 2022-05-20:
I am using the bottom launch file and I am trying to change the frequency of the (LMS5xx](https://www.sick.com/ag/en/detection-and-ranging-solutions/2d-lidar-sensors/lms5xx/c/g179651).
By default the frequency is set to 75Hz and the angular resolution is 0.5. By using the  sopas engineering  tool I was able to change the frequency to 100 Hz and 0.6667 resolution.  I would like to achieve the same result using the bottom ros launch file.
I tried to add the following lines but the scanner fail to start and the connection is closed.

```
    <param name="scan_freq" type="double" value="100" />
    <param name="ang_res" type="double" value="0.6667" /> 
```

Do u know the reason? Is it related to the client authorization mode?

**Here is a copy of the error given by the scanner.**
```
[WARN] [1653062061.401602180] [sick_scan]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0023>sMN mLMPsetscancfg 0x00 0x01 0x66 ..., received answer: "sFA\x00\x08", expected patterns: "sAN mLMPsetscancfg","sMA mLMPsetscancfg"
[WARN] [1653062066.401703512] [sick_scan]: Timeout during waiting for new datagram
[WARN] [1653062066.401847345] [sick_scan]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0023>sMN mLMPsetscancfg 0x00 0x01 0x66 ..., received answer: "", expected patterns: "sAN mLMPsetscancfg","sMA mLMPsetscancfg"
[ERROR] [1653062066.401909063] [sick_scan]: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0023>sMN mLMPsetscancfg 0x00 0x01 0x66 ..., giving up after 1 unexpected answers.
[INFO] [1653062066.401968416] [sick_scan]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[ERROR] [1653062066.402019539] [sick_scan]: ## ERROR in mainGenericLaser: init failed, retrying...
[INFO] [1653062066.402036507] [sick_scan]: Start initialising scanner [Ip: 192.168.124.101] [Port:2112]
[WARN] [1653062066.402047895] [sick_scan]: Disconnecting TCP-Connection.
[ERROR] [1653062066.402321353] [sick_scan]: Tcp::readInputData: Read 0 bytes, connection is lost!
[INFO] [1653062066.402400972] [sick_scan]: SickThread TcpRecvThread finished (flags: threadShouldRun=0, endThread=0).
sick_scan driver closed.
terminate called after throwing an instance of 'rclcpp::exceptions::ParameterAlreadyDeclaredException'
  what():  parameter 'diagnostic_updater.period' has already been declared
```

**Here there is the launch file that I am using.**
```
<launch>
    <arg name="hostname" default="192.168.124.101"/>
    <arg name="sw_pll_only_publish" default="true"/>
    <!-- launch file for the lms_5xx series -->
    <node name="sick_lms_5xx" pkg="sick_scan" type="sick_generic_caller" respawn="false" output="screen" required="true">
        <!-- launch-prefix="gdb -ex run - -args" -->

        <!-- default values: -->
        <!--
          <param name="min_ang" type="double" value="-1.658062789" />
          <param name="max_ang" type="double" value="1.658062789" />
          <param name="intensity" type="bool" value="True" />
          <param name="skip" type="int" value="0" />
          <param name="time_offset" type="double" value="-0.001" />
          <param name="publish_datagram" type="bool" value="False" />
          <param name="subscribe_datagram" type="bool" value="false" />
          <param name="device_number" type="int" value="0" />
          <param name="range_min" type="double" value="0.05" />
        -->
        <param name="frame_id" type="str" value="L1"/>
        <param name="scanner_type" type="string" value="sick_lms_5xx"/>
        <!-- -45 deg -->
        <param name="min_ang" type="double" value="-0.785398163"/>
        <!-- 45 deg -->
        <param name="max_ang" type="double" value="0.785398163"/>
        <param name="range_max" type="double" value="4.0"/>
        <param name="range_min" type="double" value="0.16"/>
        <param name="intensity" type="bool" value="true"/>
        <param name="intensity_resolution_16bit" type="bool" value="false"/>
        <param name="hostname" type="string" value="$(arg hostname)"/>
        <param name="port" type="string" value="2112"/>
        <!--<param name="timelimit" type="int" value="5"/>-->
        <param name="use_binary_protocol" type="bool" value="true"/>
        <param name="sw_pll_only_publish" type="bool" value="$(arg sw_pll_only_publish)"/>
        <param name="min_intensity" type="double" value="0.0"/> <!-- Set range of LaserScan messages to infinity, if intensity < min_intensity (default: 0) -->
        <!-- possible angular resolutions are 1,0.6667,0.5,0.3333,0.25,0.1667
         possible scan frequencies are 25,35,50,75,100 Hz
         Make sure to check in the manual whether the desired scanning frequency
         and angular resolution combination is available.
         The parameters scan_freq and ang_res must always be used together.
        -->
	      
        <param name="scan_freq" type="double" value="100" />
        <param name="ang_res" type="double" value="0.6667" /> 
      
        <param name="start_services" type="bool" value="True"/> <!-- start ros service for cola commands -->
        <param name="activate_lferec" type="bool" value="True"/> <!-- activate field monitoring by lferec messages -->
        <param name="activate_lidoutputstate" type="bool" value="True"/> <!-- activate field monitoring by lidoutputstate messages -->
        <param name="activate_lidinputstate" type="bool" value="True"/> <!-- activate field monitoring by lidinputstate messages -->

        <param name="message_monitoring_enabled" type="bool" value="True" />      <!-- Enable message monitoring with reconnect+reinit in case of timeouts, default: true -->
        <param name="read_timeout_millisec_default" type="int" value="5000"/>     <!-- 5 sec read timeout in operational mode (measurement mode), default: 5000 milliseconds -->
        <param name="read_timeout_millisec_startup" type="int" value="120000"/>   <!-- 120 sec read timeout during startup (sensor may be starting up, which can take up to 120 sec.), default: 120000 milliseconds -->
        <param name="read_timeout_millisec_kill_node" type="int" value="150000"/> <!-- 150 sec pointcloud timeout, ros node will be killed if no point cloud published within the last 150 sec., default: 150000 milliseconds -->
        <param name="client_authorization_pw" type="string" value="F4724744"/>    <!-- Default password for client authorization -->

    </node>
</launch>
```

rostest, 2022-05-23:
Thanks for reporting. We can't reproduce this error messages. sick_scan_xd and LMS 511 run normally with 100 Hz and 0.6667 resolution. Did you use the most recent sick_scan_xd version and the latest firmware?

Using different resolutions, we got no error messages according to the scan configuration with the "sMN mLMPsetscancfg" command. rviz2 shows the pointcloud as expected. Some examples just for information: 

LMS-511 configuration with 25 Hz and 0.1667:
```
        <param name="ang_res" type="double" value="0.1667" />
        <param name="scan_freq" type="double" value="25" />
```
```
[sick_generic_caller-1] [INFO] [1653294955.433996714] [sick_scan]: LaunchParser::parseFile(/home/rostest/ros_sick_scan_xd/sick_repo/install/sick_scan/share/sick_scan/launch/sick_lms_5xx.launch): name="ang_res", type="double", value="0.1667"
[sick_generic_caller-1] [INFO] [1653294955.434002263] [sick_scan]: LaunchParser::parseFile(/home/rostest/ros_sick_scan_xd/sick_repo/install/sick_scan/share/sick_scan/launch/sick_lms_5xx.launch): name="scan_freq", type="double", value="25"
[sick_generic_caller-1] [INFO] [1653294956.896638233] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x1f 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 CRC:<0x5c>
[sick_generic_caller-1] [INFO] [1653294956.897512981] [sick_scan]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[sick_generic_caller-1] [INFO] [1653294956.897595058] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[sick_generic_caller-1] [INFO] [1653294956.898475461] [sick_scan]: Receiving: <STX>sRA LMDscandatacfg \x1f\x00\x01\x00\x00\x00\x00\x00\x00\x00...
[sick_generic_caller-1] [INFO] [1653294956.898596842] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0037>sMN mLMPsetscancfg 0x00 0x00 0x09 0xc4 0x00 0x01 0x00 0x00 0x06 0x83 0xff 0xff 0x3c 0xb0 0x00 0x1c 0x3a 0x90 CRC:<0x20>
[sick_generic_caller-1] [INFO] [1653294957.052708415] [sick_scan]: Receiving: <STX>sAN mLMPsetscancfg \x00\x00\x00\x09\xc4\x00\x01\x00\x00\x06...
[sick_generic_caller-1] [INFO] [1653294957.052922741] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0014>sRN LMPscancfg CRC:<0x63>
[sick_generic_caller-1] [INFO] [1653294957.054124473] [sick_scan]: Receiving: <STX>sRA LMPscancfg \x00\x00\x09\xc4\x00\x01\x00\x00\x06\x83\xff...
```

LMS-511 configuration with 100 Hz and 0.1667:
```
        <param name="scan_freq" type="double" value="100" />
        <param name="ang_res" type="double" value="0.6667" />         
```
```
[sick_generic_caller-1] [INFO] [1653295511.666342464] [sick_scan]: LaunchParser::parseFile(/home/rostest/ros_sick_scan_xd/sick_repo/install/sick_scan/share/sick_scan/launch/sick_lms_5xx.launch): name="scan_freq", type="double", value="100"
[sick_generic_caller-1] [INFO] [1653295511.666348353] [sick_scan]: LaunchParser::parseFile(/home/rostest/ros_sick_scan_xd/sick_repo/install/sick_scan/share/sick_scan/launch/sick_lms_5xx.launch): name="ang_res", type="double", value="0.6667"
[sick_generic_caller-1] [INFO] [1653295513.129252441] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x1f 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 CRC:<0x5c>
[sick_generic_caller-1] [INFO] [1653295513.130206787] [sick_scan]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[sick_generic_caller-1] [INFO] [1653295513.130260848] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[sick_generic_caller-1] [INFO] [1653295513.131235004] [sick_scan]: Receiving: <STX>sRA LMDscandatacfg \x1f\x00\x01\x00\x00\x00\x00\x00\x00\x00...
[sick_generic_caller-1] [INFO] [1653295513.131315051] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0037>sMN mLMPsetscancfg 0x00 0x00 0x27 0x10 0x00 0x01 0x00 0x00 0x1a 0x0b 0xff 0xff 0x3c 0xb0 0x00 0x1c 0x3a 0x90 CRC:<0x4e>
[sick_generic_caller-1] [INFO] [1653295519.977715849] [sick_scan]: Receiving: <STX>sAN mLMPsetscancfg \x00\x00\x00\x27\x10\x00\x01\x00\x00\x1a...
[sick_generic_caller-1] [INFO] [1653295519.977882339] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0014>sRN LMPscancfg CRC:<0x63>
[sick_generic_caller-1] [INFO] [1653295519.979243074] [sick_scan]: Receiving: <STX>sRA LMPscancfg \x00\x00\x27\x10\x00\x01\x00\x00\x1a\x0b\xff...
```

Same result using your LMS-511 configuration as posted with 100 Hz and 0.1667:
```
        <param name="scan_freq" type="double" value="100" />
        <param name="ang_res" type="double" value="0.6667" />         
```
```
[sick_generic_caller-1] [INFO] [1653296271.244402607] [sick_scan]: LaunchParser::parseFile(/home/rostest/ros_sick_scan_xd/sick_repo/install/sick_scan/share/sick_scan/launch/sick_lms_5xx.launch): name="scan_freq", type="double", value="100"
[sick_generic_caller-1] [INFO] [1653296271.244408123] [sick_scan]: LaunchParser::parseFile(/home/rostest/ros_sick_scan_xd/sick_repo/install/sick_scan/share/sick_scan/launch/sick_lms_5xx.launch): name="ang_res", type="double", value="0.6667"
[sick_generic_caller-1] [INFO] [1653296272.707006137] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0032>sWN LMDscandatacfg 0x01 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x01 CRC:<0x42>
[sick_generic_caller-1] [INFO] [1653296272.707901651] [sick_scan]: Receiving: <STX>sWA LMDscandatacfg <ETX>
[sick_generic_caller-1] [INFO] [1653296272.707988092] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[... cut ...]
[sick_generic_caller-1] [INFO] [1653296272.829016346] [sick_scan]: Receiving: <STX>sAN mLMPsetscancfg \x00\x00\x00\x27\x10\x00\x01\x00\x00\x1a...
[sick_generic_caller-1] [INFO] [1653296272.829182448] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0014>sRN LMPscancfg CRC:<0x63>
[sick_generic_caller-1] [INFO] [1653296272.830393953] [sick_scan]: Receiving: <STX>sRA LMPscancfg \x00\x00\x27\x10\x00\x01\x00\x00\x1a\x0b\xff...
```

Please make sure using the latest sick_scan_xd version and firmware.

visiont3lab, 2022-05-23:
Thanks for the quick answer! I have update the sick_generic_caller from version 2.6.1 to 2.6.6 and everything is working fine.

closed 2022-05-23 by visiont3lab