SICKAG/sick_scan_xd #71: Timestamp investigation for TiM5xx

michael1309, 2022-06-08:
The time stamp assignment for the TiM5xx is to be investigated. If necessary, the time stamp should be selected to match the generation time of the scan.

rostest, 2022-06-20:
Until now the timestamp generation mostly references to the timestamp of the transmit timestamp.
By introducing a variabled "use_generation_timestamp" one can decide using the generation timestamp as a reference timestamp.

rostest, 2022-06-29:
Version 2.7.0 supports parameter "use_generation_timestamp":

`<param name="use_generation_timestamp" type="bool" value="true"/> `

By default, the generation timestamp of the lidarscan is converted to ros time and published. The transmit timestamp is used to adjust the software pll.

closed 2022-06-29 by rostest