SICKAG/sick_scan_xd #73: `cloud` topic uses global namespace while `scan` uses private namespace

fmessmer, 2022-06-13:
using the example launch files, e.g.
- `roslaunch sick_scan sick_nav_3xx.launch`
- `roslaunch sick_scan sick_lrs_4xxx.launch`

I found that the PointCloud2 is published on the global namespace `/cloud`
while the LaserScan is published on the privat namespace `/sick_nav_3xx/scan` or `/sick_lrs_4xxx/scan` respectively

shouldn't this be consistent?

michael1309, 2022-06-14:
As you know, publishing the pointcloud2 message is our favorite solution. 
Therefore, for harmonization, I would suggest that we also change the scan message to a private namespace. 
Would this approach be in your sense?

fmessmer, 2022-06-14:
yes...also using private namespace for the `cloud` topic as default would be most consistent - I guess.
I think all interfaces of a node should inherently be private to e.g. allow multi-sensor setup...
you can still remap to anything you like afterwards

michael1309, 2022-06-20:
The current default topic "/cloud" for pointcloud messages has been used in sick_scan, sick_scan2 and sick_scan_xd. Changing this default value requires all users to modify their configuration after an update (unless the cloud topic has been overwritten before). To support downward compatibility, we would prefer to keep this default value. 

The topic can be configured in the launchfile or by commandline. Examples for multi sensor setups are described in [https://github.com/SICKAG/sick_scan_xd#start-multiple-nodes](https://github.com/SICKAG/sick_scan_xd#start-multiple-nodes) and [https://github.com/SICKAG/sick_scan_xd/blob/master/doc/faq.md#how-to-run-multiple-sensors-concurrently](https://github.com/SICKAG/sick_scan_xd/blob/master/doc/faq.md#how-to-run-multiple-sensors-concurrently).

We only support the scan message for backward compatibility reasons. For this reason, we would like to leave the implementation in this way.

closed 2022-06-27 by michael1309