SICKAG/sick_scan_xd #79: Support for using the LiDAR LMS4XXX encoder

luangcp, 2022-06-24:
I'm having problems using the encoder on ROS1 of LiDAR Lms4xx. In sick_lms_4xxx.launch I added the parameter encoder_mode with value 2. But when I try to run the launch I find the following error:
[image]

Consulting the Telegram manual, I found that it could be a client authorization problem, but apparently everything is ok, can you help?
[image]

michael1309, 2022-06-28:
Thanks for reporting this problem. Support for encoders has been developed and tested for the LMS511 Lidar. Encoders are currently not supported by this driver for other lidars. 

The Lidar error code 0x03 means Sopas_Error_VARIABLE_UNKNOWNINDEX ("Trying to access a variable with an unknown Sopas index"), i.e. the lidar does not support the command "sWN DO3And4Fnc". According to the telegram listing, this command is specified for the LMS5xx only. Most likely, the combination of a lms4xx lidar and an encoder is not supported. 
Please ask the SICK support for further support regarding the support of an encoder. If there is a positive feedback we will discuss with the SICK management the further steps for integration of the encoder logic into the driver.

shaoshuaizyk, 2022-07-15:
> Thanks for reporting this problem. Support for encoders has been developed and tested for the LMS511 Lidar. Encoders are currently not supported by this driver for other lidars.
>
> The Lidar error [...]

I found the code sending the "sWN DO3And4Fnc" which is here: [CMD_SET_3_4_TO_ENCODER](https://github.com/SICKAG/sick_scan_xd/blob/master/driver/src/sick_scan_common.cpp#L1777#L1804). Could I just comment them out and test the encoder connecting to the LMS111 lidar? I tried, the error gone, but there's no messages showing in the encoder rostopic. I am not sure if it's the connection issue or the LMS111 does not support the encoder at all?

michael1309, 2022-07-16:
Please ask the SICK support for further support regarding the support of an encoder. If there is a positive feedback we will discuss with the SICK management the further steps for integration of the encoder logic into the driver. In principle there are two encoder inputs.

closed 2022-07-16 by michael1309