SICKAG/sick_scan_xd #88: Weird Pointcloud in mrs6124

ngifary, 2022-07-25:
I use the command
"ros2 run sick_scan sick_generic_caller ./src/sick_scan_xd/launch/sick_mrs_6xxx.launch"
to activate my sick lidar in ros2. On rviz2 I can see the published pointcloud2 and laserscan topics but when I view the pointcloud2 using the defined frame_id, I got a weird 10-sided-stars-like structure. Is this normal? I cannot recignize any object from the pointcloud.
[image]

rostest, 2022-07-26:
Thanks for reporting. The rviz-screenshot does not look normal. It might be related to the high data rate of a 24 layer scanner. Do you use the latest sick_scan_xd release? 

A record of the lidar data would be helpful for further analyzation. Can you provide recorded TCP-data from the lidar using Wireshark? If possible, please:
* start Wireshark,
* select the interface with the tcp-session to the lidar,
* start sick_generic_caller as before
* record the tcp data, and
* save it to a pcapng-file.

Please provide us with the pcapng-file for further diagnosis.

ngifary, 2022-07-27:
Yes. I use the actual sick_scan_xd from master branch.
I don't know if this is relevant but the 24 layers are somehow divided.
All 24 layers are supposed to be in the -60 to 60 degree angle but somehow layer 1,2,3, 6,7,8, 11,12,13, 16,17,18, 21,22 are in the correct angle but the rest are in the 120 to 240 degree angle. Which is why, it looks like a 10 sided stars.
In the google drive, you could find the rosbag recording as well as the wireshark recording

https://drive.google.com/drive/folders/1MSi96aN0wOCP-1iYc4aaskpuAasHqKp4?usp=sharing

rostest, 2022-07-27:
Thank you very much for the rosbag and pcapng-file! This will help a lot. I will have a closer look on this strange behaviour.

rostest, 2022-07-28:
Sorry, my fault! The elevation table has not been set correctly due to an error. The bug was introduced by version 2.7.0 (multiscan136 integration on 27 Jun 2022) and affects only MRS6124.

Thanks to your input, the bug is now fixed. I hope the following rviz-screenshot from your pcapng-file looks more familiar:
[image]

Please checkout branch https://github.com/SICKAG/sick_scan_xd/tree/feature/mrs6124-fix and rebuild sick_scan_xd. Please leave a short message about problems or sucess, so that the branch can be merged into the master trunk. Thanks again for your input!

ngifary, 2022-08-01:
Thank you very much for your quick response! I've tested it and it works fine now. I'll close this issue now.
[image]

closed 2022-08-01 by ngifary

rostest, 2022-08-01:
@ngifary Thanks for your friendly support!