SICKAG/sick_scan_xd #91: Compiled-in tiny xml

thraen, 2022-07-28:
The supplied dynamic library has tiny xml compiled in. Tiny xml symbols are visible for the dynamic linker. 

Thus, when a program links both tiny xml and sicks scan library. It may be, that the dynamic loader loads symbols from the sicks library. The program generally uses a tiny xml header whose declarations differ from the actually loaded code, leading to crashes.

I recommend hiding these (and possibly other transitively visible) symbols by means of linker options -fvisibilty=false or library version file

rostest, 2022-07-28:
@thraen Many thanks for your hint! The linker-flag "-fvisibility=hidden" is added to CMakeLists.txt. Please update and rebuild.

closed 2022-07-28 by rostest

thraen, 2022-07-28:
Thanks for the quick response!

I have to admit, I never used the visibility flag myself. I bandaged the problem downstream with a versions file for now.

With -fvisbility=hidden, don't we have to specify which symbols we want to keep visible by annotating __attribute__ ((visibility("default"))). Do I miss something?

rostest, 2022-07-29:
In my understanding, linker flag "-fvisibility=hidden" covers functions from export, i.e. you can't dynamically load functions from a shared library. The executable sick_generic_caller is linked statically and not affected. Do you have some functions in mind, which you're specifically loading dynamically?

rostest, 2022-07-29:
But I see your point. I'll remove the visibility flag for now. An API is planned for the next release. We will introduce the visibility-flag in that release.