SICKAG/sick_scan_xd #94: TIM551-205001, TCP failled to open connection

tranquil-skyflyer, 2022-08-09:
I'm using ROS2 Foxy on Ubuntu 20.04.
I have no problem connecting to the scanner via SOPAS Software but cannot seem to gain a connection with ROS2.
I've tried troubleshooting and I've double-checked the IP address.

----------
Here's what I got:

```
scanner_type                   string     sick_tim_5xx        
nodename                       string     sick_tim_5xx        
min_ang                        double     -2.35619449         
max_ang                        double     2.35619449          
use_binary_protocol            bool       true                
range_max                      double     100.0               
intensity                      bool       True                
hostname                       string     192.168.0.72        
cloud_topic                    string     cloud               
frame_id                       str        cloud               
port                           string     2112                
timelimit                      int        5                   
sw_pll_only_publish            bool       true                
use_generation_timestamp       bool       true                
min_intensity                  double     0.0                 
start_services                 bool       True                
message_monitoring_enabled     bool       True                
read_timeout_millisec_default  int        5000                
read_timeout_millisec_startup  int        120000              
read_timeout_millisec_kill_node int        150000              
client_authorization_pw        string     F4724744            
[INFO] [1660038684.587006672] [sick_scan]: Found emul_sensor overwriting default settings. Emulation:False
[INFO] [1660038684.587061758] [sick_scan]: Found sopas_protocol_type param overwriting default protocol:
[INFO] [1660038684.587093265] [sick_scan]: Binary protocol activated
[INFO] [1660038684.587190744] [sick_scan]: Start initialising scanner [Ip: 192.168.0.72] [Port:2112]
[INFO] [1660038684.588581687] [sick_scan]: Publishing laserscan-pointcloud2 to cloud
[INFO] [1660038684.588687597] [sick_scan]: Publishing on topic "cloud"
[INFO] [1660038684.589218689] [sick_scan]: Publishing on topic "sick_tim_5xx/imu"
[INFO] [1660038684.589627263] [sick_scan]: Publishing on topic "sick_tim_5xx/encoder"
[INFO] [1660038684.590053725] [sick_scan]: Publishing on topic "sick_tim_5xx/scan"
[INFO] [1660038684.590310228] [sick_scan]: sick_scan_xd: Tcp::open: connecting to 192.168.0.72:2112 ...
[ERROR] [1660038815.431804178] [sick_scan]: Tcp::open: Failed to open TCP connection to 192.168.0.72:2112, aborting.
[INFO] [1660038815.432484480] [sick_scan]: Parameter setting for <active_echo: 0>
[INFO] [1660038815.432712473] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ERROR] [1660038815.432859044] [sick_scan]: Tcp::write: Connection is not open
[ERROR] [1660038815.432915412] [sick_scan]: ## ERROR in sendSOPASCommand(): sendCommandBuffer failed
[INFO] [1660038815.432959737] [sick_scan]: Receiving: <STX><ETX>
[INFO] [1660038815.433082965] [sick_scan]: SOPAS Communication -Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>

[WARN] [1660038815.433148242] [sick_scan]: checkColaDialect: lidar response not in configured Cola-dialect Cola-B, trying different Cola configuration
[INFO] [1660038815.433182167] [sick_scan]: Sending  : <STX>sRN SCdevicestate<ETX>
[ERROR] [1660038815.433202780] [sick_scan]: Tcp::write: Connection is not open
[ERROR] [1660038815.433277155] [sick_scan]: ## ERROR in sendSOPASCommand(): sendCommandBuffer failed
[INFO] [1660038815.433306501] [sick_scan]: Receiving: <STX><ETX>
[INFO] [1660038815.433328142] [sick_scan]: SOPAS Communication -Error unexpected Sopas answer for request <STX>sRN SCdevicestate<ETX>

[WARN] [1660038815.433363123] [sick_scan]: checkColaDialect: no lidar response in any cola configuration, check lidar and network!
[WARN] [1660038815.433387035] [sick_scan]: SickScanCommon::init_scanner() failed, aborting.
[WARN] [1660038815.433407787] [sick_scan]: SickScanCommon::init_scanner(): checkColaDialect failed, restarting.
[WARN] [1660038815.433429027] [sick_scan]: It is recommended to use the binary communication (Cola-B) by:
[WARN] [1660038815.433448351] [sick_scan]: 1. setting parameter use_binary_protocol true in the launch file, and
[WARN] [1660038815.433467450] [sick_scan]: 2. setting the lidar communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
[INFO] [1660038815.433491465] [sick_scan]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[ERROR] [1660038815.433532062] [sick_scan]: ## ERROR in mainGenericLaser: init failed, retrying...
[INFO] [1660038815.433556790] [sick_scan]: Start initialising scanner [Ip: 192.168.0.72] [Port:2112]
[WARN] [1660038815.433575781] [sick_scan]: Disconnecting TCP-Connection.
sick_scan driver closed.
terminate called after throwing an instance of 'rclcpp::exceptions::ParameterAlreadyDeclaredException'
  what():  parameter 'diagnostic_updater.period' has already been declared
```

rostest, 2022-08-10:
Thanks for reporting. The driver can not open a TCP connection to 192.168.0.72 on port 2112, most probable due to network issues or firewall settings. Can you ping the lidar, `ping 192.168.0.72` ? Can you connect to the lidar with `nc -z -v -w5 192.168.0.72 2112`? See notes in https://github.com/SICKAG/sick_scan_xd/blob/master/doc/ipconfig/ipconfig.md#test-connection-under-linux and https://github.com/SICKAG/sick_scan_xd/blob/master/doc/faq.md#error-tcpopen-failed-to-open-tcp-connection-to-19216801-aborting.

TCP connections can be blocked by firewalls. See notes on https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_segment_xd.md#firewall-configuration for upd. To disable a firewall filter for 192.168.0.72:2112 on Linux, you can run
```
sudo iptables -A INPUT -p tcp -m tcp --dport 2112 -j ACCEPT
sudo iptables -A OUTPUT -p tcp -m tcp --sport 2112 -j ACCEPT
sudo iptables-save
sudo ufw allow from any to any port 2112 proto tcp
```

rostest, 2022-09-06:
Closed due to inactivity.

closed 2022-09-06 by rostest