SICKAG/sick_scan_xd #96: Publish horizontal scan for MultiScan as LaserScan-Message

NikolasE, 2022-08-26:
First of all, thanks for providing a ROS2-Interface for a prototype device! 

It would be very useful to have an additional LaserScan-Publisher for the high-resolution horizontal plane to feed this information directly into a 2D-Slam system.

michael1309, 2022-08-29:
Hello Nikolas,

actually it was decided with the SICK management not to actively support the LaserScan message anymore. 
Do you see a possibility that you can feed into the Slam algorithm the PointCloud messages? 
Do you use an open source project for SLAM-2D? 
We would be happy to support SLAM with MultiScan, of course.

NikolasE, 2022-08-29:
Hey!
Thanks for the fast answer! What was the motivation not to support LaserScan messages? 

For our experiments, I just wrote a small scripts that filters the points on the horizontal high-res plane and reconstructed a LaserScan message from it. (however, 'horizontal' meant an elevation of 1.7deg and I republished as a flat scan while assuming vertical walls). 

We are testing with open-source as well as an commercial software that all use the LaserScan message as default interface.

michael1309, 2022-08-30:
Hi Nikolas,

we got the point. We will discuss this with the SICK management again and give you a feedback until the 14th of September. We are glad that you have found a workaround in any case.

rostest, 2022-11-22:
The latest release publishes laserscan message for Multiscan lidars. By default, laserscan messages are published for layer 5 only (i.e. the layer with smallest elevation angle).  Laserscan messages for other layers can be configured by a mask using parameter laserscan_layer_filter in the launchfile:
```
    <arg name="laserscan_layer_filter" default="0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0" /> <!-- Configuration of laserscan messages (ROS only), default: laserscan messages for layer 5 activated (elevation -0.07 degree, max number of scan points) -->

        <!-- Configuration of laserscan messages (ROS only): -->
        <!-- Parameter "laserscan_layer_filter" sets a mask to create laserscan messages for configured layer (0: no laserscan message, 1: create laserscan messages for this layer) -->
        <!-- Use "0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0" to activate resp. "1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1" to activate laserscan messages for all 16 layers of the Multiscan136 -->
        <!-- Default is "0 0 0 0 0 1 0 0 0 0 0 0 0 0 0 0", i.e. laserscan messages for layer 5, (elevation -0.07 degree, max number of scan points) -->
        <param name="laserscan_layer_filter" type="string" value="$(arg laserscan_layer_filter)" />
```
Please checkout the latest release 2.8.14 and rebuild. Thanks!

closed 2022-11-22 by rostest