SICKAG/sick_scan_xd #97: cmake error at humble ros2

higherd, 2022-08-30:
Error occurs during cmake in Humble version (ros2).
I followed the README install guide but got the error below.

Every /opt/ros/humble/setup.bash , .../src/sick_scan_xd/CMakeLists.txt and ETC are installed as they are.
I couldn't find anything related to "~sick_scan_xd/include_directories" at that files.
When installing ros2 humble, I also installed ros-humble-ament-cmake*, but I don't know how to solve it.
I suspect CMakeLists.txt file rather than Cmake tool, but I don't know how to fix it.

Seek advice from experts.

> _[]/home/mercury/my_ws>colcon build --packages-select sick_scan --cmake-args " -DROS_VERSION=2" --event-handlers console_direct+
> [0.245s] WARNING:colcon.colcon_core.prefix_path.colcon:The path '/home/mercury/my_ws/sick_scan_ws/install' in the environment variable COLCON_PREFIX_PATH doesn't exist
> [0.245s] WARNING:colcon.colcon_ros.prefix_path.catkin:The path '/home/mercury/my_ws/sick_scan_ws/install/libsick_ldmrs' in the environment variable CMAKE_PREFIX_PATH doesn't exist
> Starting >>> sick_scan
> -- Option CMAKE_ENABLE_EMULATOR = , building sick_scan without emulator support
> -- Found ament_cmake: 1.3.2 (/opt/ros/humble/share/ament_cmake/cmake)
> -- Found diagnostic_updater: 3.0.0 (/opt/ros/humble/share/diagnostic_updater/cmake)
> -- Found rosidl_generator_c: 3.1.3 (/opt/ros/humble/share/rosidl_generator_c/cmake)
> -- Found rosidl_adapter: 3.1.3 (/opt/ros/humble/share/rosidl_adapter/cmake)
> -- Found rosidl_generator_cpp: 3.1.3 (/opt/ros/humble/share/rosidl_generator_cpp/cmake)
> -- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
> -- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
> -- Found rmw_implementation_cmake: 6.1.0 (/opt/ros/humble/share/rmw_implementation_cmake/cmake)
> -- Found rmw_fastrtps_cpp: 6.2.1 (/opt/ros/humble/share/rmw_fastrtps_cpp/cmake)
> -- Using RMW implementation 'rmw_fastrtps_cpp' as default
> -- Found sensor_msgs: 4.2.2 (/opt/ros/humble/share/sensor_msgs/cmake)
> -- Found nav_msgs: 4.2.2 (/opt/ros/humble/share/nav_msgs/cmake)
> -- Found tf2: 0.25.1 (/opt/ros/humble/share/tf2/cmake)
> -- Found visualization_msgs: 4.2.2 (/opt/ros/humble/share/visualization_msgs/cmake)
> -- Found tf2_ros: 0.25.1 (/opt/ros/humble/share/tf2_ros/cmake)
> -- Found rosidl_default_generators: 1.2.0 (/opt/ros/humble/share/rosidl_default_generators/cmake)
> -- Found ament_cmake_ros: 0.10.0 (/opt/ros/humble/share/ament_cmake_ros/cmake)
> -- Using all available rosidl_typesupport_c: rosidl_typesupport_fastrtps_c;rosidl_typesupport_introspection_c
> -- Using all available rosidl_typesupport_cpp: rosidl_typesupport_fastrtps_cpp;rosidl_typesupport_introspection_cpp
> -- Found python_cmake_module: 0.10.0 (/opt/ros/humble/share/python_cmake_module/cmake)
> -- Using PYTHON_EXECUTABLE: /usr/bin/python3
> -- Using PYTHON_INCLUDE_DIRS: /usr/include/python3.10
> -- Using PYTHON_LIBRARIES: /usr/lib/x86_64-linux-gnu/libpython3.10.so
> -- Building sick_scan with LDMRS support
> -- LDMRS_INCLUDES: /usr/local/lib/cmake/SickLDMRS/../../../include
> -- LDMRS_SOURCES: driver/src/ldmrs/sick_ldmrs_config.cpp;driver/src/ldmrs/sick_ldmrs_driver.cpp;driver/src/ldmrs/sick_ldmrs_node.cpp
> -- LDMRS_TARGET_DEPENDENCIES: SickLDMRS
> -- SICK_LDMRS_LIBRARIES: sick_ldmrs
> -- Building sick_scan with SCANSEGMENT_XD support
> CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
>   Use rosidl_get_typesupport_target() and target_link_libraries() instead of
>   rosidl_target_interfaces()
> Call Stack (most recent call first):
>   CMakeLists.txt:625 (rosidl_target_interfaces)
> 
> 
> CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
>   Use rosidl_get_typesupport_target() and target_link_libraries() instead of
>   rosidl_target_interfaces()
> Call Stack (most recent call first):
>   CMakeLists.txt:626 (rosidl_target_interfaces)
> 
> 
> -- Configuring done
> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
> Please set them or make sure they are set and tested correctly in the CMake files:
> /home/mercury/my_ws/src/sick_scan_xd/include_directories
>    used as include directory in directory /home/mercury/my_ws/src/sick_scan_xd
>    used as include directory in directory /home/mercury/my_ws/src/sick_scan_xd
> 
> -- Generating done
> CMake Generate step failed.  Build files cannot be regenerated correctly.
> --- stderr: sick_scan
> CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
>   Use rosidl_get_typesupport_target() and target_link_libraries() instead of
>   rosidl_target_interfaces()
> Call Stack (most recent call first):
>   CMakeLists.txt:625 (rosidl_target_interfaces)
> 
> 
> CMake Deprecation Warning at /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:32 (message):
>   Use rosidl_get_typesupport_target() and target_link_libraries() instead of
>   rosidl_target_interfaces()
> Call Stack (most recent call first):
>   CMakeLists.txt:626 (rosidl_target_interfaces)
> 
> 
> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
> Please set them or make sure they are set and tested correctly in the CMake files:
> /home/mercury/my_ws/src/sick_scan_xd/include_directories
>    used as include directory in directory /home/mercury/my_ws/src/sick_scan_xd
>    used as include directory in directory /home/mercury/my_ws/src/sick_scan_xd
> 
> CMake Generate step failed.  Build files cannot be regenerated correctly.
> ---
> Failed   <<< sick_scan [1.83s, exited with code 1]
> 
> Summary: 0 packages finished [1.97s]
>   1 package failed: sick_scan
>   1 package had stderr output: sick_scan

michael1309, 2022-08-30:
Hi, 

this is on our todo list for September 2022. 

If you are in a hurry, you can also use the pull request https://github.com/SICKAG/sick_scan_xd/pull/85 as a starting point.

higherd, 2022-08-31:
It was definitely not a problem with the environment or device installed on my PC. 
sick lidar is still in an incomplete state for ros2.
September is just around the cornet, so I guess I'll just have to wait a little while watching the current ones. Thanks

closed 2022-08-31 by higherd

higherd, 2022-08-31:
> Hi,
> 
> this is on our todo list for September 2022.
> 
> If you are in a hurry, you can also use the pull request #85 as a starting point.

The official git source is slightly(maybe) different from what you are working on, so except for some modifications (leave the argument nh of notifyPointcloudListener, in file driver/src/sick_scan_common.cpp),
And, compilation was successful by taking action only in the rosDuration part( (driver/src/ldmrs/sick_ldmrs_driver.cpp file)  where a compilation error occurred.