SICKAG/sick_scan_xd #98: Multi-Scan-Cloud contains points at origin

NikolasE, 2022-08-31:
The pointcloud2 (cloud_360) contains points at the origin, which require a filter later to make sure that there are no obstacles within the robot's footprint. I guess they could be replaced by nans or removed.

michael1309, 2022-09-01:
We will discuss this point with SICK management this month (September). I think that this value is provided directly by the firmware of the lidar. Until final clarification, we suggest that you interpret values below a certain small limit (e.g. 1 cm) as NaN.

michael1309, 2022-10-04:
We implemented the following two steps:
* Discussion with SICK management: SICK management confirmed to us that value 0 should be interpreted as an invalid measured value.
* Filter procedure: We installed filters for testing against minimum and maximum ranges in September. These changes will be online next week at the latest.

michael1309, 2022-10-10:
Using version 2.8.7 and higher you can filter against minimum range and maximum range setting. As mentioned earlier there could be range values with 0.00 indicating invalid values (see discussion above).

closed 2022-10-10 by michael1309