SICKAG/sick_scan_xd #100: Include sick_scan ros2 msg and srv inside a different ros package

visiont3lab, 2022-09-05:
Hi,
I am writing a cpp client to make a service call  that will enable and disable the laser beam.
Here is what I want to achieve

```
ros2 service call /scan3/ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sMN LMCstartmeas'}"
ros2 service call /scan3/ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sMN LMCstandby'}"
```
I would like to include this calls inside a new ros2 package called **estimator**. I have tried to to add 

```
to my CMakeLists.txt 
find_package(sick_scan REQUIRED)

to my package.xml
<depend>sick_scan</depend>
```

but I am not able to include any messages or services.

Do you build a messages/service package that I can include in my new ros package that will allow me to use sick messages and services?

rostest, 2022-09-06:
Thanks for your feedback! We will try to provide a small C++ example application using sick_scan messages and services by the end of next week.

In the meantime you could
a) build and run sick_scan_xd as a ros2 node and use system calls within your application, or
b) checkout sick_scan_xd in parallel with estimator and build. The folder structure would be e.g.
```
workspace/src/sick_scan_xd
workspace/src/estimator
```
Build in folder workspace using `colcon build --cmake-args " -DROS_VERSION=2" " -DLDMRS=0" " -DSCANSEGMENT_XD=0"` should include the sick_scan messages and services.

rostest, 2022-09-08:
We will provide a tiny ROS-2 C++ example application in the next release, which uses sick_scan messages and services. Please find a preview in the zip-file [ros2_example_application.zip](https://github.com/SICKAG/sick_scan_xd/files/9525171/ros2_example_application.zip). The zip-file contains a README.md about how to build and run a ROS-2 example application with sick_scan messages and services. Please feel free to use it as a starting point for you own sick_scan integration. Note that CMakeLists.txt has to resolve sick_scan dependencies like e.g.
 ```
 find_package(sick_scan REQUIRED)
 ament_target_dependencies(
      estimator
      "rclcpp"
      "sensor_msgs"
      "std_msgs"
      "sick_scan"
 )
 ```

michael1309, 2022-09-19:
As shown in the attached zip-file we integrated this functionality into V.2.8.4.
See https://github.com/SICKAG/sick_scan_xd/tree/master/examples/ros2_example_application as an entry point for further information.

closed 2022-09-19 by michael1309