SICKAG/sick_scan_xd #101: Using sensor sensor QOS profile.

alluring-mushroom, 2022-09-07:
We have a Linux ROS2 system with a many sources of high bandwidth data, including several SICK LMS511. We are locking down our [QOS policies.](https://docs.ros.org/en/humble/Concepts/About-Quality-of-Service-Settings.html) I have noticed that the sick driver  use the default QOS policy.

We are converting many things that don't need to be `reliable` to `best_effort`, specifically `rmw_qos_profile_sensor_data`, and I'm curious what the general opinion on using this for the SICKs is, as they seem to be what this profile targets. Specifically, it seems unlikely anyone would need to catch every single instant of the resulting pointcloud. Perhaps this is a useful change, and could be implemented as an option to keep backawards compatibility?

Thank you for your time.

rostest, 2022-09-08:
Thanks for your feedback. In general, we recommend that the network topology is designed to provide sufficient bandwidth for the transmission of the ROS messages when using the lidars.

We will include your request in our backlog and plan it for the next release. We are implementing it such that the QoS profile can be configured in the launch file. The next release is planned for the end of October 2022.

alluring-mushroom, 2022-09-09:
Thank you for your consideration and prompt response :smile:

aiplemaSICKAG, 2022-09-14:
Hi,
in the mean time you can always change the QoS on the receiving side to best_effort, which will lead to best-effort data handling, as the chosen policy is negotiated between publisher and subscriber (see the section "QoS compatibilities" on the page that you refer to: https://docs.ros.org/en/humble/Concepts/About-Quality-of-Service-Settings.html#qos-compatibilities).

Best regards,
Manuel

alluring-mushroom, 2022-09-15:
@aiplemaSICKAG
Oh, that is some useful advice. I knew they were compatible, but I assumed it would lead to the behaviour appearing to be reliable, with the publisher retrying (due to being reliable), and that not affecting the client, as it didn't care either way.

Thank you for this clarification!

closed 2022-11-22 by rostest