SICKAG/sick_scan_xd #104: Parameter change on the fly

reinbja, 2022-09-09:
Hi,
I`ve tried to change the transformation parameter on the fly but no change is visible in rviz

1. read the current parameter `rosparam get /sick_scansegment_xd/add_tranform_xyz_rpy` return =>0,0,0,0,0,0
2. set the parameter `rosparam set /sick_scansegment_xd/add_tranform_xyz_rpy "0,0,1,0,0,0"`
3. read the parameter again `rosparam get /sick_scansegment_xd/add_tranform_xyz_rpy` return =>0,0,1,0,0,0 but no change visible in rviz

If I change the parameter directly in the launch file it works perfectly.

michael1309, 2022-09-15:
After internal discussion with the SICK management we will add this specific parameter change in one of the next releases.

rostest, 2022-11-22:
An additional transform of the pointcloud can now be set during runtime using `rosparam set /sick_scansegment_xd/add_tranform_xyz_rpy`. 
Please checkout the latest release 2.8.14 and rebuild. Thanks!