SICKAG/sick_scan_xd #122: Angular Resolution LMS1xxx

christopherbohn, 2022-10-24:
Even when setting the angular resolution of the LMS1xxx in SOPAS Engineering Tool to 0.375, the /cloud is published with an angular resolution of about 0.75 degrees.

Steps to reproduce:
- Set the angular resolution in SOPAS to 0.375 & save it to the device.
- Use the sick_lms_1xxx.launch results in: **[ INFO] [1666608619.431095907]: Angle resolution of scanner is 0.75 [deg]  (in 1/10000th deg: 7500)**

The /cloud topic contains points in an angular resolution of about 0.7 degrees when analyzing them.

The expected behavior would be that the angular resolution of the /cloud topic is the one that has been set in SOPAS.

michael1309, 2022-10-25:
We took a look at this ticket and discussed it with SICK Düsseldorf. We will give you feedback in the course of the week and offer a solution directly if possible.

christopherbohn, 2022-10-25:
thx, looking forward to hearing back from you.

michael1309, 2022-10-26:
Hi Christopher,

we have updated the firmware of our device from version 1.3.2 to 1.4.1 after consultation with Kai (SICK Düsseldorf). Unfortunately, we cannot reproduce the problem this way, as there is a newer hardware revision of the device. With our device in the test, the angular resolution cannot be set. It is fixed at 1 degree. 

We will do our best to get a device with firmware version 2.x.y. Perhaps you could write us briefly which firmware version you currently see in the SOPAS ET for the device. 

We therefore ask you to be patient for a few days until we can test the behavior with a new device that we have requested from SICK.

Here is the SOPAS ET output for device. Start angle and stop angle can be modified. But this is not possible for angular resolution.
[image]

christopherbohn, 2022-10-27:
Hi Michael, Thank you for getting back. We use Version 2.2.1 - this brings "Erweiterte Einstellungen" where the frequency and angular resolution can be set to either 150 Hz - 0.75°, 75 Hz - 0.375°, or 37.5 Hz - 0.1875°.
[image]

michael1309, 2022-10-27:
Thanks for this information. It exactly matches our expectation.

christopherbohn, 2022-11-08:
Hi Michael, do you have any updates on this?

rostest, 2022-11-10:
@christopherbohn  Thanks for your patience. We could reproduce and fix the error. Please update to the new sick_scan_xd version 2.8.11.

Scan frequency and angular resolution can now be set in the new launchfile sick_lms_1xxx_v2.launch:
```
        <!--- Valid combinations of angular resolution ang_res and scan_freq for LMS-1xxx with firmware >= 2.0 are:
              150 Hz - 0.75 deg, or
              75 Hz - 0.375 deg, or
              37.5 Hz - 0.1875 deg
              Note: Configuration of scan frequency and angular resolution requires firmware version 2.0 or higher. Use sick_lms_1xxx.launch for firmware version 1.x
        -->
        <param name="ang_res" type="double" value="0.75" /> <!-- Supported values are "0.1875", "0.375" and "0.75" -->
        <param name="scan_freq" type="double" value="150" /> <!-- Supported values are "37.5", "75" and "150" -->
```

Please note that these settings require firmware version >= 2.0. For compatibility with firmware 1.x, we provide the launchfile sick_lms_1xxx.launch without this scan configuration. Please launch sick_scan_xd with sick_lms_1xxx_v2.launch, on ROS-1 e.g. by `roslaunch sick_scan sick_lms_1xxx_v2.launch`.