SICKAG/sick_scan_xd #123: Questions about mapping and navigation

hyng809, 2022-11-02:
Hello
I'm a ROS beginner and I am currently making a navigation program using SICK lidar.
Currently I'm using SICK Lidar to see Lidar working on Rviz.
But I'm stuck from this next step.
What I want to know is how to save the currently scanned map and simulate navigation using the saved map.
If you have a method or example to refer to, please let me know.
These are the hardware and software environments I am currently using.

- SICK TIM561-2050101
- Ubuntu 20.04
- ROS2 foxy

Thank you

michael1309, 2022-11-03:
Even though the question is a bit out of focus, we would like to give you the following information:

* Ready-made solution: With the SICK solutions at https://www.sick.com/no/en/localization-and-positioning-solutions/lidar-localization/lidar-loc/c/g541272 you can buy a ready-made solution.
* Open source solution: As an open source solution you could familiarize yourself with Google Cartographer. 
   * Google Cartographer: https://google-cartographer-ros.readthedocs.io/en/latest/
   * Note for the use with SICK Lidars can be found here: https://github.com/SICKAG/sick_scan_xd/blob/0c35a36cc4cf40d2a04325e2145112c27623acd1/doc/google_cartographer.md

Good luck!

hyng809, 2022-11-23:
I want to make a map using slam.
I want to know how to use it in ROS2 foxy version.
And currently /map is not available in Rviz's FIxed Frame.
(Only cloud and world are recognized.)
Do you know the solution?

[image]

[image]

aiplemaSICKAG, 2022-11-24:
Hello,
you probably want to transform the data into another frame. For this, you can use the tf node (http://wiki.ros.org/tf). However, you can best ask your questions in the ROS forum https://answers.ros.org/questions/ where experts with the ROS SLAM functionality can help you better, as the slam functionality is not provided and supported by SICK.

hyng809, 2022-11-25:
Is SLAM not available on SICK?
So, how to map using SICK lidar?

aiplemaSICKAG, 2022-11-25:
Hello,
In ROS, functionality is distributed between the different modules / nodes. The function of sick_scan_xd is to receive data from the sensor and make them available to other nodes in ROS. You can use for example hector_slam or google cartographer with the SICK ROS drivers as described here https://github.com/SICKAG/sick_scan_xd/blob/master/doc/slam.md and here https://github.com/SICKAG/sick_scan_xd/blob/master/doc/google_cartographer.md

Best regards,
Manuel

hyng809, 2022-11-28:
hello
Thank you for answer.
This is my first time using ROS and SICK LIDAR.
The two examples you sent are using ROS 1.
Currently I am using ROS2 foxy.
So I can't use the example from the site you sent me.
Do you have any way?