SICKAG/sick_scan_xd #124: setting the angle resolution RMS1000 from the launch file

Shhei, 2022-11-10:
Hello,
I want to set the angle resolution from the launch file. I have added the line <param name="ang_res" type="double" value="0.125" /> in the launch file sick_lms_1xxx.launch.  But it does not work.
Moreover, I want to scan only in one layer. What should I add to the launch file to let the device scanning in one level instead of 4 levels

rostest, 2022-11-10:
Thanks for reporting. The new sick_scan_xd version 2.8.11 released right now fixes this error.

Note, that firmware 2.x is required to set scan frequency and angular resolution. Please update to the most recent firmware version 2.2, if not yet done. Scan frequency and angular resolution can be set in the launchfile sick_lms_1xxx_v2.launch:
```
        <!--- Valid combinations of angular resolution ang_res and scan_freq for LMS-1xxx with firmware >= 2.0 are:
              150 Hz - 0.75 deg, or
              75 Hz - 0.375 deg, or
              37.5 Hz - 0.1875 deg
              Note: Configuration of scan frequency and angular resolution requires firmware version 2.0 or higher. Use sick_lms_1xxx.launch for firmware version 1.x
        -->
        <param name="ang_res" type="double" value="0.75" /> <!-- Supported values are "0.1875", "0.375" and "0.75" -->
        <param name="scan_freq" type="double" value="150" /> <!-- Supported values are "37.5", "75" and "150" -->
```
Please launch sick_scan_xd with sick_lms_1xxx_v2.launch, on ROS-1 e.g. by `roslaunch sick_scan sick_lms_1xxx_v2.launch`.  By default, a layer with 150 Hz and 0.75 degree angular resolution is published.

Shhei, 2022-11-16:
Thanks for your answer. Sorry, I made a mistake in my last message. I use the product MRS1000 and I run the launch file "sick_mrs_1xxx.launch". In this launch file I don't find the parameters ang_res or scan_freq. 
I have seen the config file SickScan.cfg, but when I change the parameter via rqt (see picture), the laser data are not affected by this and everything remained as it is, as if the values are not updated. Can you tell me, how to set the ang_res?
Should I always set the ang_res and scan_freq so that the ratio is always constant.

[image]

aiplemaSICKAG, 2022-11-24:
Hello,
can you send a copy of the console output that is produced by the driver when you change the value of ang_res? This would help us to understand the issue.
Thank you.

Best regards,
Manuel

Shhei, 2022-12-01:
Hello,
this is the console output, when I run the launch file by "roslaunch sick_scan sick_mrs_1xxx.launch". When I change the vaule of ang_res via rqt, no extra message appears in the console. I think, the problem is that the argument ang_res is defined as double and It may take a value between 0 and 10. This is physically not possible, because the angle resolution is either 0.25°, 0.125° or  0.0625°.

[image]
[image]
[image]
[image]

rostest, 2023-01-18:
Thanks for your feedback! A new sick_scan_xd version in https://github.com/SICKAG/sick_scan_xd/tree/feature/mrs1000_scan_configuration now supports configuration of the angular angle, scan frequency and layer filter. Please use this update to set the angular resolution of a MRS1000.

The angular resolution can be configured in the launchfile sick_mrs_1xxx.launch, e.g.:
```
        <param name="ang_res" type="double" value="0.25" /> <!-- Supported values are "0.0625", "0.125" and "0.25". Default: "0", i.e. do not set or overwrite angular resolution and scan frequency -->
        <param name="scan_freq" type="double" value="50" /> <!-- Supported values are "12.5", "25" and "50". Default: "0", i.e. do not set or overwrite angular resolution and scan frequency -->
        <!-- Setting a scan configuation by parameter "ang_res" and "scan_freq" requires MRS1000 firmware version >= 2.x and hardware version 2 -->
```

Setting the angular resolution requires a firmware and hardware version >= 2.x. Please update your MRS1000, if required.

Valid combinations of ang_res and scan_freq are 0.0625 / 12.5, 0.125 / 25 or 0.25 / 50. Note that the scan data and pointcloud are interlaced by factor 4 for 0.0625 deg and by factor 2 for 0.125 deg angular resolution.

Scan layer can be activated or deactivated in the launchfile by parameter "scan_layer_filter", e.g.:
```
        <!-- MRS-1000 scan layer activation mask (parameter for sopas command "sWN ScanLayerFilter"): "<number of layers> <layer 1: on/off> … <layer N: on/off>" --> 
        <!-- default: all layer activated by value="4 1 1 1 1", example: use "4 0 1 0 0" to activate layer 2 at 0 deg, use empty string "" to disable scan layer activation -->        
        <param name="scan_layer_filter" type="string" value="4 1 1 1 1" />
```
Change this filter mask to deactivate layers.

michael1309, 2023-03-23:
Closed due to inactivity.

closed 2023-03-23 by michael1309