SICKAG/sick_scan_xd #126: LMS1000 LiDAR in ROS only recording at 75hz despite explicitly being told to record at 150hz

mcguiremichael, 2022-11-15:
We are running the SICK LMS1000 lidar with ros1 melodic. We launch the sensor with the following command.

`roslaunch sick_scan sick_lms_1xxx_v2.launch hostname:=169.254.6.195 scan_freq:=150 ang_res:=0.75`

No other changes are made to the default launch file used.

This is the output from running the roslaunch command.

`
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://earthsense-Alienware-m15:42721/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /sick_lms_1xxx/add_transform_xyz_rpy: 0,0,0,0,0,0
 * /sick_lms_1xxx/ang_res: 0.75
 * /sick_lms_1xxx/client_authorization_pw: F4724744
 * /sick_lms_1xxx/cloud_topic: cloud
 * /sick_lms_1xxx/frame_id: cloud
 * /sick_lms_1xxx/hostname: 169.254.6.195
 * /sick_lms_1xxx/intensity: True
 * /sick_lms_1xxx/intensity_resolution_16bit: False
 * /sick_lms_1xxx/lfp_meanfilter: 2
 * /sick_lms_1xxx/lfp_medianfilter: 1
 * /sick_lms_1xxx/max_ang: 2.391
 * /sick_lms_1xxx/message_monitoring_enabled: True
 * /sick_lms_1xxx/min_ang: -2.391
 * /sick_lms_1xxx/min_intensity: 0.0
 * /sick_lms_1xxx/port: 2112
 * /sick_lms_1xxx/range_filter_handling: 0
 * /sick_lms_1xxx/range_max: 25.0
 * /sick_lms_1xxx/range_min: 0.0
 * /sick_lms_1xxx/read_timeout_millisec_default: 5000
 * /sick_lms_1xxx/read_timeout_millisec_kill_node: 150000
 * /sick_lms_1xxx/read_timeout_millisec_startup: 120000
 * /sick_lms_1xxx/scan_freq: 150.0
 * /sick_lms_1xxx/scanner_type: sick_lms_1xxx
 * /sick_lms_1xxx/start_services: True
 * /sick_lms_1xxx/timelimit: 5
 * /sick_lms_1xxx/use_generation_timestamp: True

NODES
  /
    sick_lms_1xxx (sick_scan/sick_generic_caller)

auto-starting new master
process[master]: started with pid [11620]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to aa4eaef0-6531-11ed-bdcd-e4b97aef6f09
process[rosout-1]: started with pid [11631]
started core service [/rosout]
process[sick_lms_1xxx-2]: started with pid [11638]
[ INFO] [1668549951.002749762]: sick_generic_caller V. 2.8.11
[ INFO] [1668549951.003419151]: Program argument 1: /home/earthsense/catkin_ws/devel_isolated/sick_scan/lib/sick_scan/sick_generic_caller
[ INFO] [1668549951.003432711]: Program argument 2: __log:=/home/earthsense/.ros/log/aa4eaef0-6531-11ed-bdcd-e4b97aef6f09/sick_lms_1xxx-2.log
[ INFO] [1668549951.003439804]: Program argument 3: __name:=sick_lms_1xxx
[ INFO] [1668549951.003448507]: ==========================================
[ INFO] [1668549951.007623361]: Range filter configuration: range_min=0, range_max=25, range_filter_handling=0
[ INFO] [1668549951.009405394]: Start initialising scanner [Ip: 169.254.6.195] [Port:2112]
[ INFO] [1668549951.037924948]: Publishing lidar pointcloud2 to cloud
[ INFO] [1668549951.037951940]: Publishing on topic "/cloud"
[ INFO] [1668549951.038304089]: Publishing on topic "/sick_lms_1xxx/imu"
[... 173 lines cut ...]
[ INFO] [1668549982.934602444]: SickScanServices: service "/sick_lms_1xxx/SickScanExit" created ("/sick_lms_1xxx/SickScanExit")
[ INFO] [1668549982.934631487]: SickScanServices: ros services initialized
[ INFO] [1668549982.934648950]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"`

After launch we recorded the cloud outputs to a rosbag. However, after running `rosbag info`, we only see 75 messages per second.

`path:        test_sick_lms1000_4_2022-11-15-16-05-53.bag
version:     2.0
duration:    1:28s (88s)
start:       Nov 15 2022 16:06:22.94 (1668549982.94)
end:         Nov 15 2022 16:07:51.18 (1668550071.18)
size:        38.3 MB
messages:    6615
compression: none [51/51 chunks]
types:       sensor_msgs/PointCloud2 [1158d486dd51d683ce2f1be655c3c181]
topics:      /cloud   6615 msgs    : sensor_msgs/PointCloud2
`

If it was recording at 150hz as we had asked it to, it would be around 13200 messages in an 88 second rosbag.

What could be causing this? The settings in SOPAS were also set to 150hz. 

Is it possible insufficient power is being provided to the sensor?

Thank you.

aiplemaSICKAG, 2022-11-24:
Hello,
thank you for reporting this. We are trying to reproduce the behavior on an equivalent sensor and find the cause of the error and will come back to you.

Best regards,
Manuel

aiplemaSICKAG, 2022-11-30:
Hello,
we think we found the source of your problem. Currently, the default launch file has the mean filter activated and configured to take the average of 2 measurements. This leads to the sensor delivering measurements at half the scan frequency. You can deactivate it by changing line 93 to `<param name="lfp_meanfilter" type="int" value="0" />`.
Note: the name of the "scan frequency" parameter is a bit misleading. With the mean filter deactivated and the output interval set to 1 in Sopas ET, scans will always be sent at 150 Hz. The difference is whether the scans are interlaced or not (see p. 22 and following in the operating instructions https://cdn.sick.com/media/docs/3/43/443/operating_instructions_lms1104c_111031s01_2d_lidar_sensors_en_im0079443.pdf ).

Best regards,
Manuel

rostest, 2023-01-09:
Thanks for your feedback! Mean and median filter settings are deactivated in default launchfile configuration in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/lms1000_configuration  and will be merged in the next release.

michael1309, 2023-03-23:
Solved due to investigation and documentation update.

closed 2023-03-23 by michael1309