SICKAG/sick_scan_xd #127: Integrate SICK scanner into C# application

PeterWimmerTmc, 2022-11-22:
Hello,

We would like to use a LMS511-20100 PRO and integrate it into our Windows C# application. We have a few questions regarding the integration:

- Is sick_scan_xd the recommended library or is there another SDK for C#?
- Where can we find binaries of sick_scan_xd (x64 DLL)?
- We plan to enable NTP in the scanner to get precise timestamps that can be synchronized with other sensors. What time base/format is the scanner using for the timestamps in the frames, how to convert them to C# DateTime structs in UTC timezone?

Any help pointing us in the right direction is appreciated.

aiplemaSICKAG, 2022-11-24:
Hello,
here some answers to your questions:

- yes, sick_scan_xd is the recommended library. However, we currently do not provide explicit support for C#
- we currently do not provide the library as pre-built binaries. But downloading the source code and building the DLL for Windows x64 should be straightforward with the instructions provided here: https://github.com/SICKAG/sick_scan_xd/blob/master/INSTALL-GENERIC.md
- the time stamps provided by the driver follow the ROS format and are provided as seconds and nanoseconds since epoch

PeterWimmerTmc, 2022-11-29:
Thank you for your explanations.

rostest, 2023-01-04:
Thanks for your feedback. I am closing the ticket for now. Please feel free to reopen the issue in case of further problems.

closed 2023-01-04 by rostest