SICKAG/sick_scan_xd #131: unexpected answer sFA\x00\x01, followed by crash

TheUncivil, 2022-11-30:
I'm currently attempting to connect to a NAV350-3232 LIDAR, launching the ROS node using the **sick_nav_3xx.launch.py** file. The LIDAR is set to use a custom IP address, which I pass to the launch file using the **hostname:=** parameter. I've set the LIDAR to use the CoLa binary protocol but am not declaring this on launch, as the ROS node seems to default to binary.

Full terminal output listed below:
`[sick_generic_caller-1] [INFO] [1669807946.068687441] [sick_scan]: Range filter configuration: range_min=0, range_max=100, range_filter_handling=0
[sick_generic_caller-1] [INFO] [1669807946.068858808] [sick_scan]: Found emul_sensor overwriting default settings. Emulation:False
[sick_generic_caller-1] [INFO] [1669807946.068912166] [sick_scan]: Found sopas_protocol_type param overwriting default protocol:
[sick_generic_caller-1] [INFO] [1669807946.068935175] [sick_scan]: Binary protocol activated
[sick_generic_caller-1] [INFO] [1669807946.069101968] [sick_scan]: Start initialising scanner [Ip: 169.254.239.240] [Port:2112]
[sick_generic_caller-1] [INFO] [1669807946.071529993] [sick_scan]: Publishing lidar pointcloud2 to cloud
[sick_generic_caller-1] [INFO] [1669807946.071590649] [sick_scan]: Publishing on topic "cloud", qos=0
[sick_generic_caller-1] [INFO] [1669807946.073684891] [sick_scan]: Publishing on topic "sick_nav_3xx/imu", qos=0
[sick_generic_caller-1] [INFO] [1669807946.074888981] [sick_scan]: Publishing on topic "sick_nav_3xx/encoder", qos=0
[sick_generic_caller-1] [INFO] [1669807946.075814047] [sick_scan]: Publishing on topic "sick_nav_3xx/scan", qos=0
[sick_generic_caller-1] [INFO] [1669807946.076204300] [sick_scan]: SickCloudTransform: add_transform_xyz_rpy = (0,0,0,0,0,0)
[sick_generic_caller-1] [INFO] [1669807946.076237038] [sick_scan]: SickCloudTransform: azimuth_offset = 0 [deg]
[sick_generic_caller-1] [INFO] [1669807946.076250288] [sick_scan]: SickCloudTransform: additional 3x3 rotation matrix = { (1,0,0), (0,1,0), (0,0,1) }
[sick_generic_caller-1] [INFO] [1669807946.076258949] [sick_scan]: SickCloudTransform: apply 3x3 rotation = false
[sick_generic_caller-1] [INFO] [1669807946.076267889] [sick_scan]: SickCloudTransform: additional translation = (0,0,0)
[sick_generic_caller-1] [INFO] [1669807946.076275480] [sick_scan]: SickCloudTransform: check_dynamic_updates = false
[sick_generic_caller-1] [INFO] [1669807946.076313508] [sick_scan]: sick_scan_xd: Tcp::open: connecting to 169.254.239.240:2112 ...
[sick_generic_caller-1] [INFO] [1669807946.077649397] [sick_scan]: sick_scan_xd Tcp::open: connected to 169.254.239.240:2112
[sick_generic_caller-1] [INFO] [1669807946.077796538] [sick_scan]: Parameter setting for <active_echo: 0>
[sick_generic_caller-1] [INFO] [1669807946.077821725] [sick_scan]: SickThread TcpRecvThread started.
[sick_generic_caller-1] [INFO] [1669807946.077867394] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[sick_generic_caller-1] [INFO] [1669807946.081358167] [sick_scan]: Receiving: <STX>sRA SCdevicestate \x00<ETX>
[sick_generic_caller-1] [INFO] [1669807946.081418954] [sick_scan]: checkColaDialect: lidar response in configured Cola-dialect Cola-B
[sick_generic_caller-1] [INFO] [1669807946.281727486] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0023>sMN SetAccessMode 0x03 0xf4 0x72 0x47 0x44 CRC:<0xb3>
[sick_generic_caller-1] [INFO] [1669807946.284924454] [sick_scan]: Receiving: <STX>sAN SetAccessMode \x01<ETX>
[sick_generic_caller-1] [INFO] [1669807946.485382586] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0015>sWN EIHstCola 0x01 CRC:<0x09>
[sick_generic_caller-1] [INFO] [1669807946.487933962] [sick_scan]: Receiving: <STX>sWA EIHstCola <ETX>
[sick_generic_caller-1] [INFO] [1669807946.688376954] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0015>sMN LMCstopmeas CRC:<0x10>
[sick_generic_caller-1] [INFO] [1669807946.691316286] [sick_scan]: Receiving: <STX>sFA\x00\x01<ETX>
[sick_generic_caller-1] [WARN] [1669807946.691563767] [sick_scan]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0015>sMN LMCstopmeas CRC:<0x10>, received answer: "sFA\x00\x01", expected patterns: "sAN LMCstopmeas","sMA LMCstopmeas"
[sick_generic_caller-1] [WARN] [1669807951.691886122] [sick_scan]: Timeout during waiting for new datagram
[sick_generic_caller-1] [WARN] [1669807951.692206134] [sick_scan]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0015>sMN LMCstopmeas CRC:<0x10>, received answer: "", expected patterns: "sAN LMCstopmeas","sMA LMCstopmeas"
[sick_generic_caller-1] [ERROR] [1669807951.692341997] [sick_scan]: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0015>sMN LMCstopmeas CRC:<0x10>, giving up after 1 unexpected answers.
[sick_generic_caller-1] [INFO] [1669807951.692411333] [sick_scan]: Failed to init scanner Error Code: 1
[sick_generic_caller-1] Waiting for timeout...
[sick_generic_caller-1] If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
[sick_generic_caller-1] This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
[sick_generic_caller-1] 1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
[sick_generic_caller-1] 2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
[sick_generic_caller-1] [ERROR] [1669807951.692464855] [sick_scan]: ## ERROR in mainGenericLaser: init failed, retrying...
[sick_generic_caller-1] [INFO] [1669807951.692544584] [sick_scan]: Start initialising scanner [Ip: 169.254.239.240] [Port:2112]
[sick_generic_caller-1] [WARN] [1669807951.692588404] [sick_scan]: Disconnecting TCP-Connection.
[sick_generic_caller-1] [ERROR] [1669807951.692778452] [sick_scan]: Tcp::readInputData: Read 0 bytes, connection is lost!
[sick_generic_caller-1] [INFO] [1669807951.692953456] [sick_scan]: SickThread TcpRecvThread finished (flags: threadShouldRun=0, endThread=0).
[sick_generic_caller-1] terminate called after throwing an instance of 'rclcpp::exceptions::ParameterAlreadyDeclaredException'
[sick_generic_caller-1]   what():  parameter 'diagnostic_updater.period' has already been declared
[ERROR] [sick_generic_caller-1]: process has died [pid 4971, exit code -6, cmd '/home/jschalkwijk/ros_ws/install/sick_scan/lib/sick_scan/sick_generic_caller /home/jschalkwijk/ros_ws/install/sick_scan/share/sick_scan/launch/sick_nav_3xx.launch hostname:=169.254.239.240 port:=2112 --ros-args'].
`

rostest, 2023-01-04:
Thanks for your feedback. The NAV350 is currently not supported by the driver. Support for NAV350 is planned within 1st quarter 2023 in one of the next releases.

closed 2023-01-04 by rostest

TheUncivil, 2023-04-21:
Do you have an update on NAV350 support? The company I work for is interested in using the NAV350 with this LIDAR.

rostest, 2023-04-24:
NAV350 support will be included in the next sick_scan_xd release. It is currently being tested and will be available in may.