SICKAG/sick_scan_xd #132: How to use sick_scan_xd_lib in QT ?

giantid, 2022-12-09:
I write some QT program code like "minimum_sick_scan_api_client.cpp" or "sick_scan_xd_api_test.cpp",  there are 3 issues :
1:  :-1: error: lidardataprocessor.o: undefined reference to symbol 'dlclose@@GLIBC_2.2.5'
2:  :-1: error: /lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
3:  :-1: error: collect2: error: ld returned 1 exit status
 what can I do ?

aiplemaSICKAG, 2022-12-09:
Hello,
can you please check that you linked against libdl? You can find an example CMakeLists here https://github.com/SICKAG/sick_scan_xd/blob/master/examples/cpp/CMakeLists.txt The important line related to your issue is `target_link_libraries(${PROJECT_NAME} PUBLIC dl) `.

Best regards,
Manuel

giantid, 2022-12-09:
>For "target_link_libraries(${PROJECT_NAME} PUBLIC dl) ", In the file “***.pro” of QT project, I added the next:
LIBS =-ldl
the first two errors disapear,only the third error remained， is it  ? 
and , I need to use  opencv, like this : 
LIBS += /usr/local/lib/libopencv_*.so
new erros come out:
 error: undefined reference to `cv::VideoWriter::VideoWriter()'
How can I do?

aiplemaSICKAG, 2022-12-09:
Hello,
that's great! The remaining error is due to the problems with linking OpenCV. Please refer to the OpenCV and QT manuals on how to build QT with OpenCV.

Best regards,
Manuel

giantid, 2022-12-09:
but the way to build OpenCV and QT is ok! 
new error is just because of the line of "LIBS =-ldl"

aiplemaSICKAG, 2022-12-09:
Hello,
be aware of the difference in the project file between "LIBS =" and "LIBS +=". The first replaces the previous content with what comes after the equal sign, the second appends it. So depending on the order in which you have the two lines "LIBS =-ldl" and "LIBS += /usr/local/lib/libopencv_.so" you might get different results. Better use "LIBS += -ldl".

Best regards,
Manuel

giantid, 2022-12-10:
Thank you very much!

rostest, 2023-01-04:
Thanks for your feedback. I am closing the ticket for now. Please feel free to reopen the issue in case of further problems.

closed 2023-01-04 by rostest