SICKAG/sick_scan_xd #133: who can explain how to set param in sick_lms_5xx.launch

giantid, 2022-12-09:
https://github.com/SICKAG/sick_scan_xd/blob/master/launch/sick_lms_5xx.launch
especially, how to set field ?  
<param name="activate_lferec" type="bool" value="True"/>
<param name="activate_lidoutputstate" type="bool" value="True"/> 
 <param name="activate_lidinputstate" type="bool" value="True"/>

aiplemaSICKAG, 2022-12-09:
Hello,
currently, the field evaluation cannot be configured from the ROS driver. Please use the SOPAS Engineering Tool (SOPAS ET) for that.

Best regards,
Manuel

giantid, 2022-12-10:
There is an error:
[Error]: cannot find parameter scanner_type in the param set. Please specify scanner_type.
but ,in the sick_lms_5xx.launch, the " scanner_type " is here:
<param name="scanner_type" type="string" value="sick_lms_5xx"/>

rostest, 2023-01-04:
Thanks for your feedback. I am closing the ticket for now. Please feel free to reopen the issue in case of further problems.

closed 2023-01-04 by rostest