SICKAG/sick_scan_xd #134: Receive messages by polling usage

nigeno, 2022-12-09:
I built and ran the sick_scan_xd_api_test.cpp example first with the default settings:
`291 bool polling = false`
it worked just fine.
Enabled the message polling option by:
`291 bool polling = true`
built ok, though when running received a Debug Assertion Failed error in Visual Studio 16.

Questions:
1. is it the proper way of switching between receiving message by registration of callbacks and by polling?
2. why receive messages by registration of callbacks is the recommended? (in the Generic API for sick_scan_xd/USAGE help)

aiplemaSICKAG, 2022-12-12:
Hello,
what was the exact message with the assert error?
1. The sick_scan_xd_api_test.cpp is a test program and while it can be very useful for inspiration, it is recommended to not modify it as is but copy snippets into other programs to avoid problems with the test setup. In your case, you are probably interested in the function whose name start with `SickScanApiWaitNext` to receive data in polling mode and the functions whose name start with `SickScanApiFree` to free data after processing.
2. Usage of callbacks is often more convenient as it will not block the main process/thread.
Best regards,
Manuel

rostest, 2023-01-04:
Thanks for your feedback. I am closing the ticket for now. Please feel free to reopen the issue in case of further problems.

closed 2023-01-04 by rostest