SICKAG/sick_scan_xd #135: Compatibility with TiM-881p?

CJdev99, 2022-12-20:
We received a TiM881 model from SiCK, and are looking to use it with ROS. Will running the 781 launch files work with the 881 model?

We had our TiM881 working fine in AppSpace, and changed the IP address to match our jetson nano's, and ran the 781 launch file. It didn't connect, and now the LiDAR won't show up in AppSpace, SOPAS, or AppManager. Any ideas for a fix?

aiplemaSICKAG, 2023-01-03:
Hello,

the TiM-881 model is not supported with ROS as this is not its designed use-case. Feel free to contact our application engineers via the SICK Support Portal https://supportportal.sick.com/ for application-specific advice and device selection. Concerning the issue to connect with AppSpace, please make sure that the computer from which you are trying to connect with the sensor has an IP-address in the same subnet as the one you configured for the sensor.

Best regards,
Manuel

rostest, 2023-01-04:
Thanks for your feedback. I am closing the ticket for now. Please feel free to reopen the issue in case of further problems.

closed 2023-01-04 by rostest