SICKAG/sick_scan_xd #137: Sick LIDAR TIM561 : Python test API won't load sick_scan_shared_lib.dll

MarinSuperbien, 2023-01-03:
Hi !
I'm trying to use a TIM561 Sick lidar with the Python test API :
[https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_api/sick_scan_api.md#complete-usage-example-in-python](https://github.com/SICKAG/sick_scan_xd/blob/master/doc/sick_scan_api/sick_scan_api.md#complete-usage-example-in-python)

I'm having trouble loading the `sick_scan_shared_lib.dll` library. When I try to import it, I get the following error message:
```python
# Load sick_scan_library
if os.name == 'nt': # Load windows dll
        sick_scan_library = SickScanApiLoadLibrary(["build/Debug/", "build_win64/Debug/", "src/build/Debug/", "src/build_win64/Debug/", "src/sick_scan_xd/build/Debug/", "src/sick_scan_xd/build_win64/Debug/", "./", "../"], "sick_scan_shared_lib.dll")

api_handle = SickScanApiCreate(sick_scan_library)
```
```
Could not find module 'sick_scan_shared_lib.dll' (or one of its dependencies). Try using the full path with constructor syntax.
```

I've checked that the library is located in a directory that is in my system's PATH environment variable, and I've also tried specifying the full path to the library when importing it. However, I'm still getting the same error message.

What could be causing this issue, and how can I fix it?
Thank you in advance for your answers.

rostest, 2023-01-04:
Thanks for your feedback. Function SickScanApiLoadLibrary loads the sick_scan_shared_lib.dll library from the list of directories given by the 1. parameter (i.e. `["build/Debug/", "build_win64/Debug/", "src/build/Debug/", "src/build_win64/Debug/", "src/sick_scan_xd/build/Debug/", "src/sick_scan_xd/build_win64/Debug/", "./", "../"]`). Please make sure that sick_scan_shared_lib.dll can be found in one of these directories. You can either copy the dll into one of the directories or adapt the directory list.

michael1309, 2023-03-23:
Closed due to inactivity.