SICKAG/sick_scan_xd #145: Qos settings and node timeout error

muttistefano, 2023-02-15:
Hi, i was using the default Qos settings, 0 in the config file, and everything worked.
I changed this to "4", to use the SensorQos, and the node crashes after 150s, giving the following error:

```
[sick_generic_caller-1] [ERROR] [1676454497.892606091] [sick_scan]: ## ERROR PointCloudMonitor: last point cloud message on topic "cloud" received 150.096 seconds ago, 150 seconds timeout exceeded.
[sick_generic_caller-1] [ERROR] [1676454497.892696062] [sick_scan]: ## ERROR PointCloudMonitor: killing process by command "nohup sleep 1 ; kill -9 1217" for restart

```
During these 150 seconds, the node works normally.
Am I doing something wrong? 
Thanks

muttistefano, 2023-02-15:
To add more info, I'm launching the node as:

```
sick = Node(
            package='sick_scan',
            executable='sick_generic_caller',
            namespace='azrael',
            output='screen',
            remappings= [('/azrael/sick_lms_1xx/scan', '/azrael/scan')],
            parameters=
            [{"intensity"                           : False},
            {"intensity_resolution_16bit"           : False},
            {"min_ang"                              : -2.35619},
            {"max_ang"                              : 2.35619},
            {"frame_id"                             :"azrael/laser"},
            {"use_binary_protocol"                  : True},
            {"scanner_type"                         :"sick_lms_1xx"},
            {"hostname"                             :"192.170.1.1"},
            {"cloud_topic"                          :"cloud"},
            {"port"                                 :"2112"},
            {"timelimit"                            : 5},
            {"min_intensity"                        : 0.0},
            {"use_generation_timestamp"             : True},
            {"range_min"                            : 0.05},
            {"range_max"                            : 25.0},
            {"scan_freq"                            : 50.0},
            {"ang_res"                              : 0.5},
            {"range_filter_handling"                : 0},
            {"add_transform_xyz_rpy"                : "0,0,0,0,0,0"},
            {"add_transform_check_dynamic_updates"  : False},
            {"start_services"                       : True},
            {"message_monitoring_enabled"           : True},
            {"read_timeout_millisec_default"        : 5000},
            {"read_timeout_millisec_startup"        : 120000},
            {"read_timeout_millisec_kill_node"      : 15000000},
            {"client_authorization_pw"              :"F4724744"},
            {"imu_enable"                           : False},
            {"ros_qos"                              : 4}])
```

I noticed that all the topics are brought under the correct namespace, all but /cloud, which is the one that triggers the timeout.
What is the purpose of this subscriber?
I also noticed that there is a mismatch of QoS in the remapped topic:

```
topic [type]            : /azrael/cloud [sensor_msgs/msg/PointCloud2]
publisher node          : sick_scan
subscriber node         : sick_scan
compatibility status    : ERROR: Best effort publisher and reliable subscription;

```
Thanks

rostest, 2023-02-15:
Thanks for reporting this issue! sick_generic_caller monitors the pointcloud messages it generates itself. Therefore, the driver subscribes to its own pointcloud topic and terminates with a timeout error, if no pointcloud messages are received within the last 150 seconds.
Just to be on the safe side: I assume the driver works as expected with the default value for parameter "ros_qos". But it fails if this (and only this) parameter is set to 4, i.e. SensorQos. Is this correct? It looks like the (internal) subscriber with default QoS does not match the publisher with SensorQos, causing the pointcloud subscription to fail. Can you post a complete log of the start sequence (output messages from logfile or screenshot)? This helps for the further diagnosis.

rostest, 2023-02-16:
We could reproduce and fix the error. The subscription to pointcloud messages in sick_generic_monitoring failed silently, if pointcloud messages are published with QoS "best effort" (SensorQoS). This caused the timeout after 150 seconds. 
The subscription now uses the configured QoS, i.e. it is identical to the QoS of the pointcloud publisher. Please use the update in branch https://github.com/SICKAG/sick_scan_xd/tree/feature/monitoring_ros2_qos and rebuild. 
Thanks again!

closed 2023-02-28 by muttistefano