SICKAG/sick_scan_xd #146: lidar watchdog timer

nigeno, 2023-02-17:
What is the best way to implement a lidar watchdog timer using the driver API?

The timer should time out in case of
1. lost TCP comms OR
2. a large number of invalid packets

The watchdog timer can be implemented in the main program using for example the callback interval, but I'm wondering if the driver can deliver directly this critical information.

Also, how can one retrieve the percentage of dropped packets from the driver?

Thank you

rostest, 2023-02-20:
The pointcloud message callback can be used as a watchdog. If no lidar messages are received for any reasons, the message callback will not be executed. Monitoring the timestamp of the last message can detect timeout and network errors.
The driver reads tcp packets with a configurable timeout (5 seconds by default) and print error messages if this timeout is exceeded. The timeout can be configured by parameter read_timeout_millisec_default in the launchfile.

nigeno, 2023-02-21:
Thank you for your reply.
Also, if the range filter is enabled, is there a way to retrieve the percentage of dropped packets from the driver? Or you have to do it in the main program.

rostest, 2023-02-23:
In principle, no TCP-packets (and therefore no ROS-messages) should be lost in TCP traffic. If you want to determine how many points are filtered by the range filter, you can examine the received point cloud and count all values that are not within the range. From the quotient of this counter and the total number of points in the point cloud you can then determine the percentage of number of valid points to all points in the pointcloud.

closed 2023-03-01 by nigeno