SICKAG/sick_scan_xd #147: sick_tim_240   Failed to start successfully

lixushi, 2023-02-26:
After executing the start command, the following error will be reported, and the topic of ros cannot be published. How to solve this problem
[image]

rostest, 2023-02-27:
Thanks for reporting. The driver can not establish a TCP connection to your Lidar using the default IP address 192.168.0.1. Reasons can be network errors (Lidar with IP-address 192.168.0.1 is unreachable), or a different IP configuration of your Lidar.

Please make sure, that the driver and your Lidar uses the same TCP subnet and that the lidar is pingable, i.e. `ping <lidar-ip>` responds without errors. If your Lidar uses another IP address, please overwrite the default via commandline using `hostname:=<ip-address>` or via launchfile.

You can find further details about the network settings and TCP errors in the FAQ: https://github.com/SICKAG/sick_scan_xd/blob/master/FAQ.md#error-tcpopen-failed-to-open-tcp-connection-to-19216801-aborting . In addition, SOPAS ET is recommended for configuration and network settings of all lidars.

michael1309, 2023-03-23:
Close due to inactivity.

closed 2023-03-23 by michael1309