SICKAG/sick_scan_xd #148: LMS511 10100 publishes topics at 250Hz

jesusramondovale, 2023-02-28:
Hi! I have a little problem with the SICK LiDAR LMS511 10100.

I've downloaded and installed (this) source repo on my Linux Ubuntu 22.04 Desktop with ROS2: Humble Hawksbill. 

These are the last 10 'echoed' lines on console (running the standard /sick_scan_ws/src/sick_scan_xd/launch/sick_lms_5xx.launch file) : 

```
[INFO] [1677594521.297062736] [sick_scan]: SickScanServices: ros services initialized
[INFO] [1677594521.297139476] [sick_scan]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"
[INFO] [1677594521.424906166] [sick_scan]: Software PLL locking started, mapping ticks to system time.
[INFO] [1677594521.425099404] [sick_scan]: 1 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1677594521.446551692] [sick_scan]: 2 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1677594521.465008859] [sick_scan]: 3 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1677594521.486231707] [sick_scan]: 4 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1677594521.505046461] [sick_scan]: 5 / 6 packets dropped. Software PLL not yet locked.
[INFO] [1677594521.526314826] [sick_scan]: Software PLL is ready and locked now!
```

Sooo... I suppose everything is working just fine: the sick_scan node starts and publishes a lot of info on multiple topics, and also, I can see the ranges' array values being published with ros2 topic echo or RQT topic monitor. 

The problem is that this /sick_scan_5xx/scan topic's being published @ 250Hz, and rviz2 doesn't paint any laser point, also, the /diagnostic topic says that "Frequency is too high". 

I've tried to change the publishing freq. on the 5xx.launch file (and re-builded) but nothing changed. It's still publishing the /scan topic at 250Hz, and I'm not able to visualize those laserp points on ROS2. 

Thanks for any help.

rostest, 2023-03-01:
Thanks for reporting this issue. We will investigate the error. Can you see the pointcloud (ros topic "cloud") in rviz at a normal frequency with default settings in the launchfile?

jesusramondovale, 2023-03-01:
Thank you for yor response!
The problem about /scan topic wasn't the frequency, it was the frame id which it was publishing on. Changed it to "map" instead of "cloud" and the Laser Scan (/scan) visualization started to go perfectly on Rviz2.

I can see the /cloud topic published by the Sick Scan node: 

[image]

As you can see, /cloud and /diagnostic topics are normally published at 100Hz, but not the /scan one

[image]

rostest, 2023-03-02:
Thanks for your feedback. We are pleased that you can now work with the scan messages. Please note that the "scan" messages are considered "legacy". We recommend working with the PointCloud2 messages. In addition, we will be looking at the deviation in the publish frequencies between /scan and /cloud in the near future.

rostest, 2023-03-06:
We tested the latest sick_scan_xd version against the LMS-511 with default settings. "ros2 topic hz /cloud" printed a publishing frequency of (approximately) 25 Hz, "ros2 topic hz /sick_lms_5xx/scan" about 36 Hz. Using a 100 Hz configuration with ang_res:=1 and scan_freq:=100, "ros2 topic hz /cloud" shows 100 Hz and "ros2 topic hz /sick_lms_5xx/scan" round about 130 Hz.

The default launchfile uses the parameter filter_echos:=1, which enables all echoes. Depending on the number of echoes received by the driver, the the published scan messages and their frequency can vary. With only one echo (i.e. filter_echos:=0 for the first echo, or filter_echos:=2 for the last echo), both /cloud and /sick_lms_5xx/scan topics are published at 100 Hz.

Please change the filter_echos setting in the launchfile to `<param name="filter_echos" type="int" value="2"/>` for the last echo, or to `<param name="filter_echos" type="int" value="0"/>` for the first echo (unless all echos are required for your application). You should then receive both cloud and scan messages at an almost constant rate of 100 Hz.

[image]

[image]