SICKAG/sick_scan_xd #149: LMS511 can not communicate using python example

arslabs, 2023-03-01:
I tried to use python example for communicating LMS511. But there may be some kind of communication error.

---
[Info]: sick_scan_api V. 2.8.15
[Info]: SickScanApiInitByLaunchfile: launchfile_args = "my.launch"
[Info]: SickScanApiInitByCli:  my.launch
[Info]: Try loading launchfile : my.launch
[Info]: LaunchParser::parseFile(my.launch): default_args["hostname"]="192.168.1.251"
[Info]: LaunchParser::parseFile(my.launch): default_args["cloud_topic"]="cloud"
[Info]: LaunchParser::parseFile(my.launch): default_args["frame_id"]="cloud"
[Info]: LaunchParser::parseFile(my.launch): default_args["sw_pll_only_publish"]="true"
[Info]: LaunchParser::parseFile(my.launch): default_args["nodename"]="sick_tim_5xx"
[Info]: LaunchParser::parseFile(my.launch): default_args["port"]="2112"
[Info]: LaunchParser::parseFile(my.launch): default_args["add_transform_xyz_rpy"]="0,0,0,0,0,0"
[Info]: LaunchParser::parseFile(my.launch): default_args["add_transform_check_dynamic_updates"]="false"
[Info]: LaunchParser::parseFile(my.launch): name="scanner_type", type="string", value="sick_tim_5xx"
[Info]: LaunchParser::parseFile(my.launch): name="nodename", type="string", value="$(arg nodename)" overwritten by default value "sick_tim_5xx"
[Info]: LaunchParser::parseFile(my.launch): name="nodename", type="string", value="sick_tim_5xx"
[Info]: LaunchParser::parseFile(my.launch): name="min_ang", type="double", value="-2.35619449"
[Info]: LaunchParser::parseFile(my.launch): name="max_ang", type="double", value="2.35619449"
[Info]: LaunchParser::parseFile(my.launch): name="use_binary_protocol", type="bool", value="true"
[Info]: LaunchParser::parseFile(my.launch): name="range_min", type="double", value="0.0"
[Info]: LaunchParser::parseFile(my.launch): name="range_max", type="double", value="100.0"
[Info]: LaunchParser::parseFile(my.launch): name="range_filter_handling", type="int", value="0"
[Info]: LaunchParser::parseFile(my.launch): name="intensity", type="bool", value="True"
[Info]: LaunchParser::parseFile(my.launch): name="hostname", type="string", value="$(arg hostname)" overwritten by default value "192.168.1.251"
[Info]: LaunchParser::parseFile(my.launch): name="hostname", type="string", value="192.168.1.251"
[Info]: LaunchParser::parseFile(my.launch): name="cloud_topic", type="string", value="$(arg cloud_topic)" overwritten by default value "cloud"
[Info]: LaunchParser::parseFile(my.launch): name="cloud_topic", type="string", value="cloud"
[Info]: LaunchParser::parseFile(my.launch): name="frame_id", type="str", value="$(arg frame_id)" overwritten by default value "cloud"
[Info]: LaunchParser::parseFile(my.launch): name="frame_id", type="str", value="cloud"
[Info]: LaunchParser::parseFile(my.launch): name="port", type="string", value="$(arg port)" overwritten by default value "2112"
[Info]: LaunchParser::parseFile(my.launch): name="port", type="string", value="2112"
[Info]: LaunchParser::parseFile(my.launch): name="timelimit", type="int", value="5"
[Info]: LaunchParser::parseFile(my.launch): name="sw_pll_only_publish", type="bool", value="$(arg sw_pll_only_publish)" overwritten by default value "true"
[Info]: LaunchParser::parseFile(my.launch): name="sw_pll_only_publish", type="bool", value="true"
[Info]: LaunchParser::parseFile(my.launch): name="use_generation_timestamp", type="bool", value="true"
[Info]: LaunchParser::parseFile(my.launch): name="min_intensity", type="double", value="0.0"
[Info]: LaunchParser::parseFile(my.launch): name="add_transform_xyz_rpy", type="string", value="$(arg add_transform_xyz_rpy)" overwritten by default value "0,0,0,0,0,0"
[Info]: LaunchParser::parseFile(my.launch): name="add_transform_xyz_rpy", type="string", value="0,0,0,0,0,0"
[Info]: LaunchParser::parseFile(my.launch): name="add_transform_check_dynamic_updates", type="bool", value="$(arg add_transform_check_dynamic_updates)" overwritten by default value "false"
[Info]: LaunchParser::parseFile(my.launch): name="add_transform_check_dynamic_updates", type="bool", value="false"
[Info]: LaunchParser::parseFile(my.launch): name="start_services", type="bool", value="True"
[Info]: LaunchParser::parseFile(my.launch): name="message_monitoring_enabled", type="bool", value="True"
[Info]: LaunchParser::parseFile(my.launch): name="read_timeout_millisec_default", type="int", value="5000"
[Info]: LaunchParser::parseFile(my.launch): name="read_timeout_millisec_startup", type="int", value="120000"
[Info]: LaunchParser::parseFile(my.launch): name="read_timeout_millisec_kill_node", type="int", value="150000"
[Info]: LaunchParser::parseFile(my.launch): name="client_authorization_pw", type="string", value="F4724744"
[Info]: LaunchParser::parseFile(my.launch): name="ros_qos", type="int", value="-1"
scanner_type                   string     sick_tim_5xx
nodename                       string     sick_tim_5xx
min_ang                        double     -2.35619449
max_ang                        double     2.35619449
use_binary_protocol            bool       true
range_min                      double     0.0
range_max                      double     100.0
range_filter_handling          int        0
intensity                      bool       True
hostname                       string     192.168.1.251
cloud_topic                    string     cloud
frame_id                       str        cloud
port                           string     2112
timelimit                      int        5
sw_pll_only_publish            bool       true
use_generation_timestamp       bool       true
min_intensity                  double     0.0
add_transform_xyz_rpy          string     0,0,0,0,0,0
add_transform_check_dynamic_updates bool       false
start_services                 bool       True
message_monitoring_enabled     bool       True
[... 100 lines cut ...]
[Info]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x13\x88\xff\xff\x3c\xb0\x00\x1c\x3a\x90<ETX>
[Info]: Angle resolution of scanner is 0.5 [deg]  (in 1/10000th deg: 5000)
[Info]: [From:To] -5 [deg] to 185 [deg] (in 1/10000th deg: from -50000 to 1850000)
[Info]: MIN_ANG: -2.35619 [rad] -135 [deg]
[Info]: MAX_ANG: 2.35619 [rad] 135 [deg]
[Info]: Sending  : <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 0x00 0x13 0x88 0xff 0xf9 0x22 0x30 0x00 0x22 0x55 0x10 CRC:<0x92>
[Info]: Receiving: <STX>sFA\x00\x04<ETX>
[Warn]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., received answer: "sFA\x00\x04", expected patterns: "sWA LMPoutputRange","sAN LMPoutputRange"
ROS_WARN: Timeout during waiting for new datagram
[Warn]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., received answer: "", expected patterns: "sWA LMPoutputRange","sAN LMPoutputRange"
[Error]: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0033>sWN LMPoutputRange 0x00 0x01 0x00 ..., giving up after 1 unexpected answers.
[Info]: Failed to init scanner Error Code: 1
Waiting for timeout...
If the communication mode set in the scanner memory is different from that used by the driver, the scanner's communication mode is changed.
This requires a restart of the TCP-IP connection, which can extend the start time by up to 30 seconds. There are two ways to prevent this:
1. [Recommended] Set the communication mode with the SOPAS ET software to binary and save this setting in the scanner's EEPROM.
2. Use the parameter "use_binary_protocol" to overwrite the default settings of the driver.
## ERROR in mainGenericLaser: init failed, retrying...
ROS_WARN: Disconnecting TCP-Connection.
Tcp::readInputData: Read 0 bytes, connection is lost!
sick_generic_laser: leaving main loop...[Info]: SickThread TcpRecvThread finished (flags: threadShouldRun=0, endThread=0).
sick_scan driver closed.

--
I used sick_tim_5xx.launch launch file.

rostest, 2023-03-01:
Thanks for reporting this issue. For communication with a LMS5111, you have to use the launchfile sick_lms_5xx.launch. Please use the local file sick_scan_xd/launch/sick_lms_5xx.launch or https://github.com/SICKAG/sick_scan_xd/blob/master/launch/sick_lms_5xx.launch to run the python example in combination with a LMS511.

michael1309, 2023-03-23:
Close due to inactivity.