SICKAG/sick_scan_xd #150: LMS4XXX: how to get intensity value

tianyuan07, 2023-03-06:
My device is LMS4124R, I want to get intensity value, I set intensity true in the launch file,like this:
 <param name="intensity" type="bool" value="True"/>
but I got all points' intensity  value 0, I wonder what's the range of intensity value, and how can I get it ?

aiplemaSICKAG, 2023-03-07:
Hello,
please make sure that you enabled the RSSI channel in SOPAS ET by checking the checkbox in Application -> Output data format. If that does not solve the issue, please contact the SICK support portal: https://supportportal.sick.com/

[image]

Regards,
Manuel

aiplemaSICKAG, 2023-03-07:
The intensity values should be approximately in the range 0..1000.

closed 2023-03-07 by aiplemaSICKAG