SICKAG/sick_scan_xd #152: How do I build sick_scan with option SCANSEGMENT_XD_SUPPORT?

chance2006, 2023-03-12:
Dear Sir:
I tested multiscan&Picoscan(Timtwo), according the guidance "Build on Linux ROS1".
I finished all steps，with the exception of creating source code documentation by d oxygen 
I always get error:sick_tim two not supported,please build sick_scan with option SCANSEGMENT_XD_SUPPORT when I run: roslaunch sick_scan sick_tim two.launch.

Could you kindly give me some guidance? Thank you very much.

BR.
Chester Hao

aiplemaSICKAG, 2023-03-13:
Hello,

please refer to this branch, which build with scansegment_xd support enabled: https://github.com/SICKAG/sick_scan_xd/tree/feature/scansegment_xd_support

Best regards,
Manuel

michael1309, 2023-03-14:
Hello,

the TimTwo is currently not supported. For the MRS136 (multiscan) we retried the steps given in the installation documentation and it works.

We recommend in any case to cleanup the following directory before building:
```
rm -rf ./build ./build_isolated/ ./devel ./devel_isolated/ ./install ./install_isolated/ ./log/
```

michael1309, 2023-03-14:
In any case we updated the documentation regarding the build steps a little bit in this branch.

aiplemaSICKAG, 2023-03-14:
Hello,

the issue was with options being overwritten by values in the CMake cache if cmake was called more than once. With the latest CMakeLists.txt from the feature/scansegment_xd_support branch, this issue should be solved. Please make sure to use that version ( https://github.com/SICKAG/sick_scan_xd/blob/feature/scansegment_xd_support/CMakeLists.txt ) and build with the following command:
```
catkin_make_isolated --install --cmake-args -DROS_VERSION=1 -DSCANSEGMENT_XD=1 -Wno-dev
```
We will apply the changes to the next regular release.

Best regards,
Manuel

closed 2023-03-23 by michael1309