SICKAG/sick_scan_xd #156: [sick_scan_xd with C#] Driver for C#

toan70411hd91, 2023-03-21:
Hello everyone,

Wish you a great day and hope you have time to help me for this question.

I am working on a project with Sick 2D LiDARs (on Windows, using C#). My problem is that I cannot find out the way to build a driver for Sick 2D LiDARs to use with C#. I have tried to download this package, then in Windows Terminal:
- mkdir build
- cd build
- cmake .. -DBUILD_CSHARP_BINDINGS=ON -DBUILD_SHARED_LIBS=ON

=> Warning:
"CMake Warning:
  Manually-specified variables were not used by the project:
    BUILD_CSHARP_BINDINGS"
 
It seems this package does not support for C#. Can you please give me some guides to use Sick driver with C#?

Thank you so much and hope to hear from you soon,

Notes:
- Before, I did the same way for camera and I worked. So, I tried to do same, but this package seems not support for C#
  
- I used SICK 2D LiDARs (TiM, LMS, microScan and nanoScan) with ROS, and they worked all great.

michael1309, 2023-03-23:
A direct C# connection is currently not provided. We recommend using the generic library for C. Here you can create a wrapper that provides the data for C#.

toan70411hd91, 2023-03-23:
@michael1309 : Thank you so much for your reply. So, you meant that I can create a C# wrapper from "sick_scan_xd"? Can you please explain a bit clearer?