SICKAG/sick_scan_xd #157: Debian releases for Foxy and Humble

tonybaltovski, 2023-03-22:
Can you please do a debian release for Foxy and Humble? Like #36.

michael1309, 2023-03-23:
Thank you for your feedback. A Debian release is on our long term todo list and will most likely be implemented in the second half of the year.

tonybaltovski, 2023-03-25:
@michael1309 can this issue be left open till the release is made?

michael1309, 2023-03-27:
Yes. That is ok for us.

obadul024, 2023-04-23:
I am trying to install on ROS2 Humble and I get an error that says;
```
sick_scan_ws/src/sick_scan_xd/include/sick_scan/ldmrs/sick_ldmrs_driver.hpp:58:10: fatal error: sick_ldmrs/manager.hpp: No such file or directory
   58 | #include <sick_ldmrs/manager.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated
```

Is this error related to Humble Release?
Thanks

rostest, 2023-04-24:
For LDMRS devices libsick_ldmrs is required. Checkout https://github.com/SICKAG/libsick_ldmrs.git and follow the instructions on https://github.com/SICKAG/sick_scan_xd/blob/master/INSTALL-ROS2.md:
```
mkdir ./src
pushd ./src
git clone https://github.com/SICKAG/libsick_ldmrs.git
git clone https://github.com/SICKAG/msgpack11.git
git clone https://github.com/SICKAG/sick_scan_xd.git
popd
colcon build --packages-select libsick_ldmrs --event-handlers console_direct+
source ./install/setup.bash
colcon build --packages-select msgpack11 --cmake-args " -DMSGPACK11_BUILD_TESTS=0" --event-handlers console_direct+
colcon build --packages-select sick_scan --cmake-args " -DROS_VERSION=2" --event-handlers console_direct+
source ./install/setup.bash
```
If you do not need or want to support LDMRS, you can switch off option BUILD_WITH_LDMRS_SUPPORT in [CMakeLists.txt](https://github.com/SICKAG/sick_scan_xd/blob/master/CMakeLists.txt) or call colcon with option -DLDMRS=0:

```
colcon build --packages-select sick_scan --cmake-args " -DROS_VERSION=2" " -DLDMRS=0" --event-handlers console_direct+
```

messageid, 2023-06-01:
> I am trying to install on ROS2 Humble, Linux Ubuntu 22.04, and I get a compilation error that terminates the build. Error is as follows;
> 
> ```
> sick_scan_ws/src/sick_scan_xd/include/sick_scan/ldmrs/sick_ldmrs_driver.hpp:58:10: fatal error: sick_ldmrs/manager.hpp: No such file or directory
>    58 | #include <sick_ldmrs/manager.hpp>
>       |          ^~~~~~~~~~~~~~~~~~~~~~~~
> compilation terminated
> ```
> 
> Is this error related to Humble Release? Thanks

Even disabling BUILD_WITH_LDMRS_SUPPORT, colcon build still report: fatal error: sick_ldmrs/manager.hpp: No such file or directory.

I've succefull compile sick_scan on ROS2 Humble modifying sick_scan_xd CMakeLists.txt on line 26 from:

 set(BUILD_WITH_LDMRS_SUPPORT ON CACHE BOOL "Build ${PROJECT_NAME} with LDMRS support (ON, libsick_ldmrs from https://github.com/SICKAG/libsick_ldmrs required) or without LDMRS support (OFF)" FORCE)

to 

 set(BUILD_WITH_LDMRS_SUPPORT OFF CACHE BOOL "Build ${PROJECT_NAME} with LDMRS support (ON, libsick_ldmrs from https://github.com/SICKAG/libsick_ldmrs required) or without LDMRS support (OFF)" FORCE)

I'm not sure if this is a bug of CMakeLists or not.

rostest, 2023-06-05:
Please run the following steps to build sick_scan_xd without LDMRS support on Linux using ROS-2 Humble:

```
mkdir -p ./sick_scan_xd_ws/src
cd ./sick_scan_xd_ws/src
git clone https://github.com/SICKAG/msgpack11.git
git clone https://github.com/SICKAG/sick_scan_xd.git
cd ..
source /opt/ros/humble/setup.bash
colcon build --packages-select msgpack11 --cmake-args " -DMSGPACK11_BUILD_TESTS=0" --event-handlers console_direct+
colcon build --packages-select sick_scan --cmake-args " -DROS_VERSION=2" " -DLDMRS=0" --event-handlers console_direct+
source ./install/setup.bash
```

Alternatively (i.e. instead of cmake option `" -DLDMRS=0"`), you can switch off option `BUILD_WITH_LDMRS_SUPPORT` in CMakeLists.txt line 25. See https://github.com/SICKAG/sick_scan_xd/blob/master/INSTALL-ROS2.md for further build options and instructions.

Please run `rm -rf ./build/ ./install/ ./log/` to ensure a complete rebuild of sick_scan_xd.

rostest, 2024-02-13:
Prebuilt sick_scan_xd binaries for ROS-2 humble can be installed by `sudo apt-get install ros-humble-sick-scan-xd`.

closed 2024-02-13 by rostest