SICKAG/sick_scan_xd #158: Redirecting std::cout causes driver failure

nigeno, 2023-03-28:
Why when trying to redirect messages from driver console to a GUI interface, it causes the driver loosing connection?
```
ERROR in sendSOPASCommand(): sendCommandBuffer failed
ERROR in sendSOPASCommand(): sendCommandBuffer failed
ERROR in mainGenericLaser: init failed, retrying...
ROS_WARN: Disconnecting TCP-Connection.
```
What is the best way to redirect driver console output without affecting it?
Thanks

michael1309, 2023-03-29:
The ROS sample application is intended as a console application. Therefore, a redirection to the GUI is not actually intended. If you try it anyway, it should actually have no effect on the interaction between Lidar and the Linux system.
Maybe it would be an alternative for you to use the generic version of the driver which works without ROS. Then you could embed the functionality well into a GUI based program.

michael1309, 2023-03-29:
Maybe it could also be helpful that you redirect the output to files and then read them in the GUI.

Informational link: https://medium.com/linuxstories/bash-pipes-and-redirections-4c267c13643b

nigeno, 2023-03-29:
Thank you for your reply. Running the generic version on Windows. Redirecting streams with redbuf() function led to the driver connection failure even when running on a different thread. As for redirection to files, I tried it too, the same affect, but didn't try it on a different thread. BTW attaching a new windows console to the gui application and redirecting output stream to it leads to the same failure. That's why I was wondering if it could potentially affect the driver, don't have much time to analyze the code.

closed 2023-04-05 by nigeno