SICKAG/sick_scan_xd #160: Cannot stop and start measurment with LD-OEM1500

pinkponk, 2023-04-08:
I know the LD-OEM1500 is not in the list of supported devices but it works when I run 
```bash
ros2 run sick_scan sick_generic_caller ./src/sick_scan_xd/launch/sick_nav_3xx.launch
```
However I cannot get the services to work correctly. I want to be able to start and stop the lidar measurments using:
```bash
ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sMN LMCstopmeas'}"
ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sMN LMCstartmeas'}"
```

But nothing happens the the service server seems to freeze. I get the following outputs when for example tying to stop:
```bash
[INFO] [1680947349.517571246] [sick_scan]: SickScanServices: Sending request "sMN LMCstopmeas"
[INFO] [1680947349.517643292] [sick_scan]: Sending  : <STX><STX><STX><STX><Len=0015>sMN LMCstopmeas CRC:<0x10>
[INFO] [1680947349.518777915] [sick_scan]: Receiving: <STX>sFA\x00\x01<ETX>
[WARN] [1680947349.518823912] [sick_scan]: Error Sopas answer mismatch: Error unexpected Sopas answer for request <STX><STX><STX><STX><Len=0015>sMN LMCstopmeas CRC:<0x10>, received answer: "sFA\x00\x01", expected patterns: "sAN LMCstopmeas","sMA LMCstopmeas"
```

If I send more commands I no longer get the prints.

I have run with the windows program, connected, started, stopped using the gui. everything worked But when I try using the terminal in the gui program and sending the exact same commands as I see when I use the buttons it still dosent work. What am I doing wrong?

I also could not find the "lidar control" as this issue spoke about: https://github.com/SICKAG/sick_scan_xd/issues/95

Note: this works
```bash
ros2 service call /SickScanExit sick_scan/srv/SickScanExitSrv "{}"
```
but I dont want to kill the node just pause it.

michael1309, 2023-04-09:
Thanks for your feedback. Error code 0x01 ("Receiving: <STX>sFA\x00\x01<ETX>") means "wrong userlevel, access to method not allowed". You can try these commands as a authorized client using 
```
ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sMN SetAccessMode 3 F4724744'}"
ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sMN LMCstopmeas'}"
ros2 service call /ColaMsg sick_scan/srv/ColaMsgSrv "{request: 'sMN LMCstartmeas'}"
```

Please note, that sick_generic_caller monitors the measurement telegrams from the lidar. By default, sick_generic_caller reinitializes both the TCP-connection and the lidar after a timeout of 150 seconds. To pause sick_generic_caller for a longer time, parameter read_timeout_millisec_kill_node must be increased in the launchfile, e.g. to 24 hours (86400000 ms):
```
        <param name="read_timeout_millisec_kill_node" type="int" value="86400000"/> <!-- 86400000 sec pointcloud timeout, ros node will be killed if no point cloud published within the last 150 sec., default: 150000 milliseconds -->
```

Additionally, we recommend to use launchfile sick_oem_15xx.launch, i.e. to run `ros2 run sick_scan sick_generic_caller ./src/sick_scan_xd/launch/sick_oem_15xx.launch`.

pinkponk, 2023-04-10:
Thank you for the answer!

I got it working with the commands you gave me. I was sure I had tried setting the access mode but I guess not :+1: 

I had to increase the `read_timeout_millisec_default` in addition to `read_timeout_millisec_kill_node` you mentioned.

--- 
Another question:

In the launch file there is this line:
`<param name="scan_cfg_list_entry" type="int" value="1"/>`
Is this mode the same as the:
```xml
<!--
        |Mode |Inter-laced |Scan freq. | Result. scan freq.| Reso-lution |Total Resol. | Field of view| Sector| LRS 3601 3611 |OEM 1501|NAV 310 |LRS 3600 3610 |OEM 1500|
        |   |   |       |        |        |         |       |                 |   |   |   |   |   |
        |1  |0x |8 Hz   |8 Hz    |0.25°   |0.25°    |360°   |0 ...  360°      |x  |x  |x  |(x)|(x)|
        |2  |0x |15  Hz |15  Hz  |0.5°    |0.5°     |360°   |0 ...  360°      |x  |x  |x  |(x)|(x)|
        |3  |0x |10  Hz |10  Hz  |0.25°   |0.25°    |300°   |30  ... 330°     |x  |x  |x  |x  |x  |
        |4  |0x |5 Hz   |5 Hz    |0.125°  |0.125°   |300°   |30  ... 330°     |x  |x  |x  |x  |x  |
        |5  |0x |6 Hz   |6 Hz    |0.1875° |0.1875°  |360°   |0 ...  360°      |x  |x  |x  |(x)|(x)|
        |6  |0x |8 Hz   |8 Hz    |0.25°   |0.25°    |359.5° |0.25° ...359.25° |   |   |   |x  |X  |
        |8  |0x |15  Hz |15  Hz  |0.375°  |0.375°   |300°   |30...330°        |x  |X  |x  |x  |x  |
        |9  |0x |15  Hz |15  Hz  |0.5°    |0.5°     |359°   |0.5  ... 359.5°  |   |   |   |x  |x  |
        |21 |0x |20  Hz |20  Hz  |0.5°    |0.5°     |300°   |30  ... 330°     |   |X  |x  |   |x  |
        |22 |0x |20  Hz |20  Hz  |0.75°   |0.75°    |360°   |0 ...  360°      |   |x  |x  |   |(x)|
        |44 |4x |10  Hz |2.5  Hz |0.25°   |0.0625°  |300°   |30  ... 330°     |x  |x  |   |(x)|(x)|
        |46 |4x |16  Hz |4 Hz    |0.5°    |0.125°   |300°   |30  ... 330°     |   |x  |   |   |(x)|
         -->
```
And is changing the mode this way still unsupported as the comment says?

closed 2023-04-10 by pinkponk

michael1309, 2023-04-11:
Great that everything has now worked for you. 
Mode 1 refers to this table. That is correct. We recommend to use mode 1. We have found that changing modes otherwise can lead to non-regular states. Therefore, if there is no explicit reason to use the other modes on your side, we recommend using mode 1. However, the parsing logic within the driver generally supports other parameter settings as well.