SICKAG/sick_scan_xd #161: ROS_INFO: Software PLL is ready and locked now!

ouyangziyao, 2023-04-13:
Hello, sir. I run the sick_lms_5xx.launch on sick Ims511 on Windows with no ros. But I can't receive the Complete point cloud coordinate information.  but there are part coordinates.Could you help me?

This is my sick_Ims_5xx.launch file.
[image]

This is my Log information.

F:\Lidar_Demo\sick_scan_ws\sick_scan_xd>.\build\Debug\sick_scan_xd_api_test.exe launch/sick_lms_5xx.launch hostname:=192.168.0.12

sick_scan_xd_api_test started
[Info]: sick_scan_api V. 2.8.15
[Info]: SickScanApiInitByCli: .\build\Debug\sick_scan_xd_api_test.exe launch/sick_lms_5xx.launch hostname:=192.168.0.12
[Info]: Try loading launchfile : launch/sick_lms_5xx.launch
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): default_args["hostname"]="192.168.0.10"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): default_args["cloud_topic"]="cloud"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): default_args["frame_id"]="cloud"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): default_args["sw_pll_only_publish"]="true"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): default_args["nodename"]="sick_lms_5xx"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): default_args["add_transform_xyz_rpy"]="0,0,0,0,0,0"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): default_args["add_transform_check_dynamic_updates"]="false"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="min_ang", type="double", value="-1.658062789"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="max_ang", type="double", value="1.658062789"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="cloud_topic", type="string", value="$(arg cloud_topic)" overwritten by default value "cloud"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="cloud_topic", type="string", value="cloud"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="frame_id", type="str", value="$(arg frame_id)" overwritten by default value "cloud"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="frame_id", type="str", value="cloud"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="scanner_type", type="string", value="sick_lms_5xx"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="range_min", type="double", value="0.05"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="range_max", type="double", value="100.0"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="range_filter_handling", type="int", value="0"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="intensity", type="bool", value="true"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="intensity_resolution_16bit", type="bool", value="false"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="hostname", type="string", value="$(arg hostname)" overwritten by default value "192.168.0.10"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="hostname", type="string", value="192.168.0.10"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="port", type="string", value="2112"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="timelimit", type="int", value="5"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="use_binary_protocol", type="bool", value="true"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="filter_echos", type="int", value="1"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="sw_pll_only_publish", type="bool", value="$(arg sw_pll_only_publish)" overwritten by default value "true"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="sw_pll_only_publish", type="bool", value="true"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="use_generation_timestamp", type="bool", value="true"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="min_intensity", type="double", value="0.0"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="add_transform_xyz_rpy", type="string", value="$(arg add_transform_xyz_rpy)" overwritten by default value "0,0,0,0,0,0"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="add_transform_xyz_rpy", type="string", value="0,0,0,0,0,0"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="add_transform_check_dynamic_updates", type="bool", value="$(arg add_transform_check_dynamic_updates)" overwritten by default value "false"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="add_transform_check_dynamic_updates", type="bool", value="false"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="start_services", type="bool", value="True"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="activate_lferec", type="bool", value="True"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="activate_lidoutputstate", type="bool", value="True"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="activate_lidinputstate", type="bool", value="True"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="message_monitoring_enabled", type="bool", value="True"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="read_timeout_millisec_default", type="int", value="5000"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="read_timeout_millisec_startup", type="int", value="120000"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="read_timeout_millisec_kill_node", type="int", value="150000"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="client_authorization_pw", type="string", value="F4724744"
[Info]: LaunchParser::parseFile(launch/sick_lms_5xx.launch): name="ros_qos", type="int", value="-1"
min_ang                        double     -1.658062789
max_ang                        double     1.658062789
cloud_topic                    string     cloud
frame_id                       str        cloud
scanner_type                   string     sick_lms_5xx
range_min                      double     0.05
range_max                      double     100.0
range_filter_handling          int        0
intensity                      bool       true
intensity_resolution_16bit     bool       false
hostname                       string     192.168.0.10
port                           string     2112
timelimit                      int        5
use_binary_protocol            bool       true
[... 206 lines cut ...]
[Info]: apiTestVisualizationMarkerMsgCallback(apiHandle:0000023FB32A5260): VisualizationMarker message, 12 objects
[Info]: apiTestLIDoutputstateMsgCallback(apiHandle:0000023FB32A5260): LIDoutputstate message, state=(1,1,1,1,1,1,0,0), count=(1,11,11,11,11,11,0,0)
[Info]: apiTestVisualizationMarkerMsgCallback(apiHandle:0000023FB32A5260): VisualizationMarker message, 12 objects
[Info]: apiTestLIDoutputstateMsgCallback(apiHandle:0000023FB32A5260): LIDoutputstate message, state=(1,1,1,1,1,1,0,0), count=(1,11,11,11,11,11,0,0)
[Info]: apiTestVisualizationMarkerMsgCallback(apiHandle:0000023FB32A5260): VisualizationMarker message, 12 objects
[Info]: 2 / 6 packets dropped. Software PLL not yet locked.
[Info]: 3 / 6 packets dropped. Software PLL not yet locked.
[Info]: 4 / 6 packets dropped. Software PLL not yet locked.
[Info]: 5 / 6 packets dropped. Software PLL not yet locked.
ROS_INFO: Software PLL is ready and locked now!
[Info]: apiTestPolarPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestCartesianPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestPolarPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestCartesianPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestPolarPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestCartesianPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestPolarPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestCartesianPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestPolarPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[... cut ...]
[Info]: apiTestPolarPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestCartesianPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestPolarPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestCartesianPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestPolarPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...

ouyangziyao, 2023-04-13:
This is sick Ims 5xx.launch file. I see you answer about min_angle and max_angle from Google. Is there a error in launch file?

[image]

[image]

rostest, 2023-04-17:
Thank you for reporting this. Please use parameter min_ang und max_ang from the default launchfile for LMS 5xx (i.e. sick_lms_5xx.launch):
```
        <param name="min_ang" type="double" value="-1.658062789" />
        <param name="max_ang" type="double" value="1.658062789" />
```

The safety field configuration with 699 points in field 22 looks weird. We recommend to configure all safety fields with SOPAS ET. Please post all log messages (i.e. the complete startup sequence) in case of further problems.

rostest, 2023-04-18:
Thanks for your reply and logfiles. The logfile MyLog.txt in the zipfile shows that pointcloud messages are received, i.e. the callback for pointcloud messages is executed:
```
[Info]: apiTestPolarPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestCartesianPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestPolarPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
[Info]: apiTestCartesianPointCloudMsgCallback(apiHandle:0000023FB32A5260): 381x5 pointcloud callback...
```
This indicates that the driver is working as expected. To view the coordinates of the lidar data, you need to convert the pointcloud message for further processing. Use the conversion functions in files sick_scan_xd_api_test.cpp, sick_scan_api_converter.cpp and api_impl.cpp as an example, e.g. functions plotPointcloudToJpeg, SickScanApiConverter::convertPointCloudMsg or convertPointCloudMsg for processing of pointcloud messages received in apiTestCartesianPointCloudMsgCallback. 

Note that sick_scan_api_test.cpp is just an example about how to receive and process pointcloud and other messages. It uses function plotPointcloudToJpeg to plot the coordinates into a jpg-image. Function plotPointcloudToJpeg in sick_scan_xd_api_test.cpp shows how to convert a pointcloud message to cartesian coordinates. We recommend to use this function as a starting point for your own data processing.

When viewing laserscan messages in rviz, the view can flicker, since the lidar generates multiple layers for each elevation angle, but laserscan messages support only one layer. You can avoid flickering by increasing the decay time in rviz. But we recommend to use pointcloud messages, because they contain a full scan of all layers. Laserscan messages are mostly supported for downward compatibility.

The safety field configuration with 699 points in field 22 looks strange. We recommend to use SOPAS ET for configuration.

ouyangziyao, 2023-04-19:
Thanks you. Is there a sample program for measuring distance with Lms511? Can I use the formula that is sqrt(x²+y²) to measuring the objects are the front of Lms511  after I set Lms511 to (0,0)?

ouyangziyao, 2023-04-22:
I'm sorry to bother you once again. 
I have added some of my own code to this program. But on the CMD, the program keeps stopping at "[ INFO] [1682040974.301047169]: 1 / 6 packets dropped.
[ INFO] [1682040974.317386553]: 2 / 6 packets dropped. software PLL not yet locked.
[ INFO] [1682040974.336678858]: 3 / 6 packets dropped. software PLL not yet locked.
[ INFO] [1682040974.357351262]: 4 / 6 packets dropped. software PLL not yet locked.
[ INFO] [1682040974.376593204]: 5 / 6 packets dropped. software PLL not yet locked.". I don't know what this log means? The code I added contains some information output, but the command line does not show the information I added. Could you tell me what this log means? I can't find the answer in the search engine. This file is a log message I generated with ros1 on ubuntu. rviz can display the point cloud message normally. After that, I also recorded some pcd files using ros.

 
Closed #161 as completed.

michael1309, 2023-04-23:
The so-called software PLL is used to synchronize the so-called ticks from the lidar to the host system time. At the beginning, values are collected in the software PLL until enough data are collected for the alignment (typically 6 or 7). This is indicated by the message 
```
<index> / 6 packets dropped. software PLL not yet locked.
```

This behavior and display is correct and intentional. You can read more information here:
https://github.com/SICKAG/sick_scan_xd/blob/master/doc/software_pll.md

In particular you can find
https://github.com/SICKAG/sick_scan_xd/blob/master/doc/software_pll.md#how-software-pll-works
the explanation, why here at the beginning is waited.