SICKAG/sick_scan_xd #164: Error :sick-scansegment xd not supported, please build sick scan with option SCANSECMENT_XD_SUPPORT

linjx0422, 2023-04-24:
Dear Sir:
I tested multiscan136, 
According the guidance "Build on Windows".
I follow the guidance documents，when i run commend “.\build\Debug\sick_scan_xd_api_test.exe launch/sick_lms_5xx.launch hostname:=192.168.0.1“.
Error occurs:"sick-scansegment xd not supported, please build sick scan with option SCANSECMENT_XD_SUPPORT"

same time I also test at Linux ROS1
 when I run:" roslaunch sick_scan sick_scansegment_xd.launch hostname:=192.168.0.1 udp_receiver_ip:=192.168.0.100 "
same Error occurs:"sick-scansegment xd not supported, please build sick scan with option SCANSECMENT_XD_SUPPORT"

I don't konw why

Could you kindly give me some guidance? Thank you very much.

rostest, 2023-04-24:
Option SCANSEGMENT_XD support is not activated by default. Please use branch https://github.com/SICKAG/sick_scan_xd/tree/feature/scansegment_xd_support, which activates SCANSEGMENT_XD support by default.
Alternatively, you can build sick_scan_xd with cmake option `-DSCANSEGMENT_XD=1` after building the msgpack11 library: 
`cmake -DROS_VERSION=0 -DSCANSEGMENT_XD=1 -G "%_cmake_string%" ..` on Windows resp. 
`catkin_make_isolated --install --cmake-args -DROS_VERSION=1 -DSCANSEGMENT_XD=0 -Wno-dev` on Linux/ROS1.
See also https://github.com/SICKAG/sick_scan_xd/issues/152 and https://github.com/SICKAG/sick_scan_xd/issues/163#issuecomment-1519631597 for further information.
To ensure a complete rebuild, please remove all files in the build-folder and rebuild.

sjeffreydong, 2023-04-24:
	Testing: multiScan136 
	Testing systems: windows 11 visual studio 2019 community 
	Testing results:

run: cmake -DROS_VERSION=0 -G "%_cmake_string%" ..

Result: "sick-scansegment xd not supported, please build sick scan with option SCANSECMENT_XD_SUPPORT"
[image]

what is reason caused this situation？

rostest, 2023-04-24:
Please use branch https://github.com/SICKAG/sick_scan_xd/tree/feature/scansegment_xd_support, which activates SCANSEGMENT_XD support by default, e.g. by `git clone --branch feature/scansegment_xd_support https://github.com/SICKAG/sick_scan_xd.git`.
Alternatively, you can build sick_scan_xd with cmake option `-DSCANSEGMENT_XD=1` after building the msgpack11 library: 
`cmake -DROS_VERSION=0 -DSCANSEGMENT_XD=1 -G "%_cmake_string%" ..` (i.e. add `-DSCANSEGMENT_XD=1` in the cmake calle).
To ensure a complete rebuild, please remove all files in the build-folder and rebuild.

linjx0422, 2023-04-25:
hello，thank your reply。
I try to use command：”cmake -DROS_VERSION=0 -DSCANSEGMENT_XD=1 -G "%_cmake_string%" to rebuild ，but the problem still exist！  I don‘t know why
[image]

rostest, 2023-04-25:
Thanks for reporting. Unfortunately, I can not reproduce this error. The error message looks strange, since tmpmaxang is declared above in line 465:

[image]

Can you check lines 465 to 486 in rossimu.cpp and make sure they are identical to the screenshot? Do you use Visual Studio 2019? In this case, can you open the visual studio project file `sick_scan.sln` in the build folder with Visual Studio 2019 and rebuild within the IDE? 

Please post a screenshot of the error message and a complete logfile of all build messages, if building still fails.

linjx0422, 2023-04-25:
> 
Thanks for your help, I remove all files and rebuild again, this time there is no error!

closed 2023-04-25 by linjx0422